betaflight/betaflight #5642: Parameter groups not resetting properly when updating

mikeller, 2018-04-07:
Steps to reproduce:
- build SPRACINGF3 target from 31639e4a4; install with 'full chip erase';
- use `diff all` to get clean diff, save to file;
- build SPRACINGF3 target from 9a7533f57; install without 'full chip erase';
- use `diff all` to get clean diff, save to file;
- compare the two files.

Expected result: The files are identical.

Result: See diff listing below.

Hypothesis: The actual length of parameter groups is not properly persisted / taken into account when reading them, and the length of the parameter group that is stored in the firmware (and may be larger than what it was when the parameter group was stored with an older version of the firmware) is used, causing all following parameter groups to be read with an offset.

```
$ diff -u BTFL_cli_20180407_130201.txt BTFL_cli_20180407_130435.txt
--- BTFL_cli_20180407_130201.txt	2018-04-07 13:02:18.696539717 +1200
+++ BTFL_cli_20180407_130435.txt	2018-04-07 13:04:43.187484610 +1200
@@ -1,5 +1,5 @@
 diff all
-# Betaflight / SPRACINGF3 (SRF3) 3.4.0 Apr  7 2018 / 12:59:00 (31639e4a4) MSP API: 1.39
+# Betaflight / SPRACINGF3 (SRF3) 3.4.0 Apr  7 2018 / 13:02:32 (9a7533f57) MSP API: 1.39
 defaults nosave
 
 feature -TRANSPONDER
@@ -12,16 +12,76 @@
 profile 0
 rateprofile 0
 
+set throttle_limit_type = PROSHOT1000
+set throttle_limit_percent = 0
 rateprofile 1
 
+set thr_mid = 0
+set thr_expo = 100
+set rates_type = DEFAULT
+set pitch_rc_rate = 0
+set yaw_rc_rate = 0
+set pitch_expo = 70
+set yaw_expo = 70
+set pitch_srate = 10
+set yaw_srate = 0
+set tpa_rate = 114
+set tpa_breakpoint = 50
 rateprofile 2
 
+set thr_mid = 100
+set thr_expo = 100
+set roll_rc_rate = 0
+set pitch_rc_rate = 0
+set yaw_rc_rate = 70
+set roll_expo = 70
+set pitch_expo = 70
+set yaw_expo = 10
+set roll_srate = 0
+set pitch_srate = 114
+set yaw_srate = 6
+set tpa_rate = 50
+set tpa_breakpoint = 25600
+set throttle_limit_type = ALWAYS
 rateprofile 3
 
+set thr_mid = 0
+set roll_rc_rate = 70
+set pitch_rc_rate = 70
+set yaw_rc_rate = 70
+set roll_expo = 10
+set yaw_expo = 114
+set roll_srate = 6
+set pitch_srate = 50
+set yaw_srate = 0
+set tpa_rate = 0
+set tpa_breakpoint = 25700
+set throttle_limit_percent = 0
 rateprofile 4
 
+set thr_mid = 0
+set thr_expo = 70
+set rates_type = ICM20608G
+set roll_rc_rate = 70
+set pitch_rc_rate = 10
+set yaw_rc_rate = 0
+set roll_expo = 114
+set pitch_expo = 6
+set yaw_expo = 50
+set roll_srate = 0
+set pitch_srate = 0
+set yaw_srate = 100
+set tpa_rate = 100
+set tpa_breakpoint = 0
+set throttle_limit_percent = 70
 rateprofile 5
 
+set thr_mid = 70
+set thr_expo = 70
+set rates_type = I2C
+set roll_rc_rate = 0
+set pitch_rc_rate = 114
+set yaw_rc_rate = 6
 rateprofile 0
 save
 # 
\ No newline at end of file
```

etracer65, 2018-04-07:
Well maybe this is a special case.  For the later commit the extra parameters were added to settings but no change to the actual PG was made since the elements already existed.  Maybe this is a case where we need to update the PG version even though it didn't actually change.  But not 100% that would have prevented this issue.

etracer65, 2018-04-07:
Also the elements being pointed to in settings.c were the first element of an array or structures so maybe that messed up the offset?  Although `vbat_scale` was already previously there doing the same thing - just pointing to the first byte of the array.

mikeller, 2018-04-07:
I have not yet done an in-depth analysis. The above is just a clean-room reproduction of a problem I ran into when updating the firmware on a craft, and it shows that even when everything is done according to the process we're currently using it can still break. It could be that this is an edge case because the new elements are in an array of structs, but past user reports suggest it's also happening in cases where the new elements aren't in an array.

Some approaches we should look into in order to address and fix this:
1. detect if a parameter group in flash is shorter than what the firmware expects. This needs to work, and work reliably;
2. if 1. is detected, zero pad the missing part of the config in memory. This might have unintended consequences if the default for the missing parameters is not zero;
**or**
3. if 1. is detected, reset the parameter group in question. This will potentially lead to the user missing the fact that _some_ parameters have been reset, while the large majority has been preserved throughout the firmware upgrade;
**or**
4. if 1. is detected, reset all parameter groups. This is drastic, but it will prevent compromised configurations from being flown;
**or**
5. (a variant of 3.) if 1. is detected, reset the parameter group in question, and set a `config_incomplete` parameter. Disable arming if `config_incomplete` is set, requiring the user to fix the config and reset `config_incomplete`.

etracer65, 2018-04-07:
On option 3 we discussed earlier and I seem to remember coming to the conclusion that option 4 was better (albeit more drastic).  With option 3 how would the user know which items were reset and which were left alone?  There suggestion was to have a popup when the user connects notifying him that the settings have been reset.

I'm not sure about having to reset a `config_imcomplete` flag as it sounds like a huge opportunity for support issues - "Just upgraded and can't arm..."

ledvinap, 2018-04-16:
`controlRateConfig_t` is array, PGs don't handle element size change (yet) ..

mikeller, 2018-04-16:
@ledvinap: Thanks, we figured that out in the meantime. iNav is discussing ways on how to solve this, we'll look into following suit.

ledvinap, 2018-04-16:
@mikeller : 'chained' PG header may be quite elegant fix - configRecord_t.flags may contain bit that internal structure data follow, first bytes of  `uint8_t pg[];` will encode type + length + another continuation bit of internal structure representation ('array + 1 additional byte + no continuation / item size' in current implementation) ..

mikeller, 2018-04-16:
@ledvinap: That sounds like a good idea.

@alberto: Does iNav have the parameter group / size discussion in a place where we could participate?

mikeller, 2018-06-02:
@alberto: Has anything been done about this in iNav? If possible I'd like to get this fixed (and do so in a way that stays compatible with iNav) for 3.4.

mikeller, 2018-06-10:
Fixed in #5991.

closed 2018-06-10 by mikeller