betaflight/betaflight #5652: RevoltOSD does not support LED's and DShot at the same time

randyfox69, 2018-04-08:
Steps to reproduce:

1.  DFU flash
2. Set ESC output to DSHOT (600)
3. Enable LED's

Observe motors 3 + 4 do not spin. Both motors are viewable and editable in BLHeliSuite, but do nothing in either the configurator or when arming manually.

I can provide a dump if needed, but the steps to reproduce are easy enough.

I assume it is some sort of shared timer issue?

mikeller, 2018-04-08:
@randyfox69: Can you please provide a `diff`, `resource list`, and `dma list`, for a config with Dshot enabled, in a version with LED_STRIP enabled and one without LED_STRIP enabled?

DieHertz, 2018-04-08:
We know about this issue and know how to solve it supposedly, I'll submit a PR tomorrow

randyfox69, 2018-04-09:
@DieHertz  thanks!

@mikeller  if still required I will, though it sounds like @DieHertz  knows what the issue is. Maybe he can elaborate.

DieHertz, 2018-04-10:
@randyfox69 Please try this build to see if it works.
[betaflight_3.3.1_REVOLTOSD.hex.zip](https://github.com/betaflight/betaflight/files/1893115/betaflight_3.3.1_REVOLTOSD.hex.zip)

randyfox69, 2018-04-10:
@DieHertz  I can confirm it works!

PR #5665 (2018-04-10): Fix timer definition for REVOLTOSD
Fixes #5652, REVOLTOSD had suboptimal timer arrangement with TIM5_UP DMA colliding to TIM4_CH1 DMA.
Also motor 3 & 4 had pin names mixed up, but it did not matter as we don't use them directly for motor output.
closed 2018-04-10 by mikeller