betaflight/betaflight #5657: Instability with D setpoint transition at near 0 (0.01, 0.03)  3.3.1, 3.4.0

flint723, 2018-04-09:
I've seen it reported more than once now and experienced it once myself.  I didn't have time to do any more testing yet but will try to repeat with 3.3.0, 3.3.1, as well as 3.4.0 nightly.

For me, I experienced the issue when setting D setpoint weight to 1.0 and transition to 0.01 on 3.4.0 build 719 from master.  I don't normally fly these settings so have nothing to compare yet, but another user reported that this worked in 3.3.0 but was unstable in 3.3.1.  When I flew it I immediately saw huge instability on roll axis and landed (see log).  Unfortunately I did not save diff before changing settings (and later, firmware), or get better logs (which is why I didn't post this sooner).  The flight right after this I set 0.0 transition and 0.7 weight, no issues at all.

[image]
Log:
[flight 178 throttle boost, rc interp, setpt weight 1, transition 0,01.zip](https://github.com/betaflight/betaflight/files/1888489/flight.178.throttle.boost.rc.interp.setpt.weight.1.transition.0.01.zip)

In this image you see rc command is nowhere near zero (and relatively stable) therefore transition should not be having any effect.  Roll P is going nuts, Roll D isn't working as it should.

I hope to get more logs in the next day or so but thought I would post this now.

Maybe has something to do with the fairly recent "discontinuity fix" #5509 or "dterm low transition value glitch fix" #5484.

DieHertz, 2018-04-09:
Since transition is a reciprocal value, it's not a surprise that `1/x` is discontinuous at 0.

flint723, 2018-04-09:
Are you saying this is expected behavior now?  Because it (apparently) was not in 3.3.0.

DieHertz, 2018-04-09:
I don't think it was ever different from `1/x`, maybe something else has changed

flint723, 2018-04-09:
Need more data, I will get it.

richard-scott, 2018-04-09:
Not sure if this is exactly the same, but I had a similar issue where I thought D setpoint weight/transition was causing sluggish responses by the craft...

Turns out hovering with Airmode on is a bad idea as it locks attitude too soon for hovering in a confined space e.g a 2m square box.

The setpoint options just made this more apparent.

flint723, 2018-04-09:
I'm not sure that's related, what I'm talking about here is large uncommanded movement/instability.  Not sluggish response to control inputs.

flint723, 2018-04-09:
Ok, the issue definitely exists.  So far in 3.3.0 I can set 1.0 (also 1.5) weight, 0.01 transition with no bad behavior.  

In 3.3.1 I started to see the issue (first noticed a slight jerk when entering a flip/roll) at 0.1 transition, and it gets progressively worse as I set 0.05, then 0.01.  At 0.01 transition the quad accelerates **_to the point of activating gyro overflow protection_** when entering a flip or roll.  I have DVR of this flight (was changing transition in OSD) and the behavior is pretty spectacular.  Will link it here when it uploads.

Here is the 3.3.1 log and diff, the sessions in this log are as follows, Transition 0.2, 0.1, 0.2, 0.05, 0.01
[flight 196 3.3.1 d setpt wht 1,0 transition 0.2, 0.1, 0.2, 0.05, 0.01.zip](https://github.com/betaflight/betaflight/files/1891239/flight.196.3.3.1.d.setpt.wht.1.0.transition.0.2.0.1.0.2.0.05.0.01.zip)
[diff 3.3.1 d transition issue.txt](https://github.com/betaflight/betaflight/files/1891242/diff.3.3.1.d.transition.issue.txt)

DieHertz, 2018-04-09:
Transition of 0 disables transition altogether, it was a PR by @ctzsnooze I think

flint723, 2018-04-09:
I don't know why the behavior of transition would have changed from 3.3.0 to 3.3.1.  That's why I suggested the PR "discontinuity fix" #5509 might have something to do with it.

If this is actually correct behavior (in 3.3.1 and 3.3.4) with low transition values then we need to not allow these near zero values to be set.

ctzsnooze, 2018-04-09:
I looked closely at the 'discontinuity fix' code and can't see how it would cause any specific issue that would affect your log.  This code applies whenever (a) weight is not zero, and (b) all the time if transition = 0 and not when your sticks are right in the middle when transition is low.   I've had zero issues in normal flight on 3.4 with weight = 1 and transition = 0.  I very much doubt that there is any problem arising from the 'discontinuity fix' itself, around transition point or otherwise.  The discontinuity fix and just separated out the application of the derivative maths on gyro separately from setpoint, which is the way it should have been done originally.  I very much doubt any issue with this code at all.

Transition code has not changed in a long time.  The glitch issue I identified that applies when transition is set to low values.  That glitch can only happen when a stick crosses the centre point of stick travel.  It would show up as just one glitch in PID_D while accelerating a stick input across the transition zone.   This is not what your log shows.  It shows a sort of haywire behaviour, like bad prop-wash instability.  But this is happening while none of the sticks are in the transition zone.  

My assessment is that neither code change could explain what happens in this log.

To sort of prove that this issue exists would require some careful testing of transition 0 vs 0.01, comparing 3.4 nightly to 3.3.1 say.  There may be some specific bug in 3.3.1 causing this issue.

At the same time, I think that extremely low transition values serve no useful purpose and should not be permitted.  Even a 10% wide transition zone seems relatively pointlessly narrow to me.  Why someone wants D weight with all it's benefits to disappear when making small around-centre inputs beats me.  But anyway I'd support changing the minimum value for transition to 10 (0.1) just because lower values will cause a small glitch on crossing centre, at least in theory.

ctzsnooze, 2018-04-09:
And yes, transition is a 1/x calculation, that's why previously 0 was disallowed.  But 1/0.01 is perfectly OK, or should be.  

My PR simply made certain that setting transition to 0 disables the transition maths completely, and thereby allowed entering transition of 0 and removing all transition effects.  So comparing transition=0 to some other value in 3.4 builds should be useful for testing purposes.

flint723, 2018-04-10:
Okay, if there is nothing obvious there, I guess I'll just skip back randomly through the 3.4 nightlies.  See if I can narrow it down.  I might not have time for this for a couple days though.

I also have no issue with 0 transition (in 3.3.1 or 3.4.0), only near zero.  I agree the minimum value for transition should be limited to 10-20 now that 0 can be set.

flint723, 2018-04-14:
Finally got to testing this on 3.4 nightlies.  I can say with confidence it was the "discontinuity fix" merge that caused this issue.  No idea why, I am not smart enough to understand what was changed.  

Flew 3.4.0 #699 and #700 from master back to back (#700 was discontinuity fix merge) and 699 had no hint of the issue, 700 it is there.  I have logs, diff, and DVR.

#699:
[3.4 #699 no issue dvr002.zip](https://github.com/betaflight/betaflight/files/1911052/3.4.699.no.issue.dvr002.zip)
[diff 3.4 #699.txt](https://github.com/betaflight/betaflight/files/1911053/diff.3.4.699.txt)

#700:
[3.4 #700 has issue dvr001.zip](https://github.com/betaflight/betaflight/files/1911050/3.4.700.has.issue.dvr001.zip)
[diff 3.4 #700.txt](https://github.com/betaflight/betaflight/files/1911051/diff.3.4.700.txt)

I can also provide the DVR files to sync to BB.

It would be nice if someone else could fly these and verify.  But be aware the issue can be pretty extreme (in this test when I rolled, it did 5+ rolls before I regained control).

DieHertz, 2018-04-14:
I'll give you a new build to try, would you be able to?

DieHertz, 2018-04-14:
[betaflight_3.4.0_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/1911066/betaflight_3.4.0_OMNIBUSF4.hex.zip)
@flint723 there you go

flint723, 2018-04-14:
Yes

flint723, 2018-04-14:
@DieHertz the attached build does not have the issue.

DieHertz, 2018-04-14:
Thanks, I'll submit a PR.

flint723, 2018-04-16:
Is it just a revert? Sorry just curious.

DieHertz, 2018-04-16:
Nope, it's not, I forgot to commit before pushing. Fixed now :-)

DieHertz, 2018-04-16:
Please try the new build at `ci.betaflight.com` when it completes, after you report back we can close this issue :-)
It'll likely be available no sooner than [Build 766](https://ci.betaflight.tech/job/Betaflight/766)

DieHertz, 2018-04-16:
Reopening until @flint723 provides feedback

flint723, 2018-04-16:
Yes, I've flown build 766, it is fixed.

closed 2018-04-16 by flint723

m013, 2018-05-08:
so, there is no fix for 3.3.2? 
[image]

matek f405aio

[LOG00002.zip](https://github.com/betaflight/betaflight/files/1985857/LOG00002.zip)

flint723, 2018-05-08:
The fix for this issue should be in 3.3.2 even if not listed on the main "features" page.  I will look at your log in a moment.

mikeller, 2018-05-08:
@m013: The fix is in 3.3.2.

flint723, 2018-05-08:
I see from the log that the dterm instability has persisted with near zero Dterm transition.  I would suggest just putting that to zero instead of 0.01 (in cli if you do not use 10.3.0 configurator).

I don't know why the fix has't fixed this for you, can you test if it is also a problem in 3.4 CI builds?

m013, 2018-05-09:
@flint723 i wanted to try  3.4's the new d term filters a week ago or so on my another quad with emax magnum f4 (obf4 target), but osd was all messed up there again(as far as i remember it hasn't been fixed since march) so i didnt think of doing it on matek.

the issue here seems mitigated at transition 0.6 and more.

i ve just flashed 3.3.0 and everything went back to normal

mikeller, 2018-05-09:
Sorry, I spoke too soon. #5701 was marked for inclusion in 3.3.2, but was not included. Sorry for that, it will be in 3.3.3.

flint723, 2018-05-09:
@m013 Also I use obf4 target with 3.4 (but not on Magnum) and haven't seen messed up osd for a long time.  So if Magnum in particular has an issue then that board must not be compatible with obf4 target.  Have you created an issue here for the osd problem?

m013, 2018-05-12:
@flint723 no , i didnt, but on the other quad with obf4 target it looks fine
but i ve got another issue.
https://github.com/betaflight/betaflight/issues/5876

m013, 2018-06-02:
it is still present in 3.3.3 , caught on another quad
[image]
[BTFL_BLACKBOX_LOG_Hawk5_20180602_183129_bad bearing_333.zip](https://github.com/betaflight/betaflight/files/2065349/BTFL_BLACKBOX_LOG_Hawk5_20180602_183129_bad.bearing_333.zip)

m013, 2018-06-02:
@mikeller , but there are the same spikes on the graph like  from the other quad's log above ( 3.3.2)
[LOG00005.zip](https://github.com/betaflight/betaflight/files/2065447/LOG00005.zip)
[image]

PR #5701 (2018-04-16): Constrained D-term transition upper-bound
Fixed #5657. Looks like it was overlooked originally in #5509.