betaflight/betaflight #5660: VTX Mod to change the channel - Problem

arierep60, 2018-04-09:
I guys, I made this VTX MOD so I can use a switch on my radio to change channel, power, etc but now I have a problem...

https://www.youtube.com/watch?v=0Xhzg45I36k

I used 20x20mm Matek F411 on my 3" quad and a Airbot F4 clone on my 5" quad.
The mod works fine on both drones and I can change everything on vtx now with a switch but everytime I power on my drone it moves on to the next vtx channel automatically, EVERY SINGLE TIME, so, when I want to fly I have to go for all the channels so I can get mine again :(
I already disabled all beeper settings in betaflight but this keep happening, but when I have my fc connected to the betaflight and plug in a battery it doesn't do that.

Update: Today I made a few tests with a friend and now we know that the problem is on the FC's

He got 4 drones with this mode, all different FC's and only 3 work fine, the 4th doesn't and have the same issue as mine have, once he plug in the battery the vtx jump to the next channel.

This happens with or without the radion turned on so the problem is in the fc/betafight configuration.

We used VTX03 on every quad, but not the same vtx, 1 on each quad.

Any sugestion for a CLI command or something to make this work on any fc? Because I bet this could be fixed anyway but I don't know how :(

Thanks

mikeller, 2018-04-09:
This issue is not with the beeper (since you are removing the beeper functionality from the beeper pin entirely, before assigning the pinio function to it), but with the fact that different pins on the MCU will have different output configurations on power up, before they get initialised. This means that you will or will not get the 'phantom button push' on power up depending on which pin is used for the beeper on the board you are using.

You could make this a bit broader applicable by not using the output driver connected to the beeper pin, but adding your own output driver to any spare pin on the board. Potentially it will even be possible to drive the VTX push button input directly without a driver.

Good idea for a mod by the way!

arierep60, 2018-04-10:
Thanks for your reply mikeller ;)
So, for now the best thing to do is just wait  for some "improvements" for this new features?

mikeller, 2018-04-10:
@arierep60: As I said in my previous post, the problem that you are experiencing is a hardware problem, so it's not something that can be 'fixed' or 'improved'. Testing what pins can be used that don't generate the spike on reboot will help, but it will have to be done for each individual target.

arierep60, 2018-04-10:
I will leave this as it was and have to keep using my vtx button with a toothpick ahahah.
Thank you for your help my friend ;)

closed 2018-04-11 by arierep60