betaflight/betaflight #5664: "Random" when Airmode is on, motors turn full throttle

scheri05, 2018-04-10:
Everything is new in this Setup! 

FC: Fortini F4 OSD, ESC: Hobbywing 4-1 40A Dshot600
I have three exactly the same Quads

The issue occurs on all quads:

[I tried it in Betaflight 3.2.4, 3.3, 3.4 stable versions] If i turn on Airmode (on Switch or as feature, tried both) and arm, the motors turn instant full throttle. If i disable Airmode the problem doesnt happen. Doesnt matter what i configure, it happens also with the standard betaflight setup.
It only occurs sometimes. One time i flashed the new FC it worked normal with airmode ON. The next day i tried again and the problem was there. It happens Random. Last Weekend i was flying and it worked with Airmode ON. The motors not went full throttle arming. Suddenly with a new battery i wanted to start and the quad liftoff with full throttle...
At this point i turned off Airmode and had no problems the whole day.

Cant figure out what the problem is. Spent over 8hours testing all possible settings...

Maybe someone can help me out with this issue. Thank you for reading this

etracer65, 2018-04-11:
It sounds like you have something wrong with your configuration.  It may help if you porst a `diff` as requested in the issue creation guidelines.  I would recommend doing a full chip erase when flashing and manually configuring your settings (don't past an old dump or diff).

scheri05, 2018-04-13:
the problem was my rc...

closed 2018-04-13 by scheri05