betaflight/betaflight #5669: FW version BF3.3.1 and above sBUS broke

Ozzy33, 2018-04-11:
At FW version 3.3.1 and above sBUS broke, vervious version worked.
Hardware: X4R-SB, Revolt FC
I have green light on the x4r, but no Rx in BFC and wont arm.
Arming disable flags: RXLOSS CLI MSP

I tried:
3.2.5 = OK
3.3.0 = OK
3.3.1 = No Good
3.4.0-755 = No Good

Version
# Betaflight / REVOLT (RVLT) 3.4.0 Apr 9 2018 / 21:32:39 (696478d04) MSP API: 1.39
betaflight_3.4.0_REVOLT-755-696478d04.hex

Status
System Uptime: 30 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=240MHz, Vref=3.32V, Core temp=61degC, GYRO=ICM20602, ACC=ICM20602
Stack size: 2048, Stack address: 0x1000fff0
I2C Errors: 1, config size: 2293, max available config: 16384
CPU:33%, cycle time: 31, GYRO rate: 32258, RX rate: 49, System rate: 9
Arming disable flags: RXLOSS CLI MSP

Please let me know if you want anything tested, thanks!

mikeller, 2018-04-11:
@Ozzy33: In general, when reporting a bug, attaching a `diff` will help others help you.

My suspicion is that what you've got is the same as https://github.com/betaflight/betaflight/pull/5647#issuecomment-379509509

Ozzy33, 2018-04-11:
Thanks Mike, diff and status dump

`# Betaflight / REVOLT (RVLT) 3.4.0 Apr  9 2018 / 21:32:39 (696478d04) MSP API: 1.39
# feature
feature AIRMODE
# aux
aux 0 0 0 900 1600 0
aux 1 26 0 900 1200 0
# master
set gyro_overflow_detect = YAW
set gyro_use_32khz = ON
set acc_calibration = 27,-63,-22
set min_check = 1015
set rc_interp = MANUAL
set rc_interp_int = 13
set blackbox_device = SPIFLASH
set min_throttle = 1055
set motor_pwm_protocol = MULTISHOT
set small_angle = 180
set pid_process_denom = 1
set debug_mode = GYRO_RAW
set cpu_overclock = 240MHZ

set setpoint_relax_ratio = 30
set dterm_setpoint_weight = 200
set iterm_rotation = ON

set roll_rc_rate = 249
set pitch_rc_rate = 249
set yaw_rc_rate = 229
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 10
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0`

`# version
# Betaflight / REVOLT (RVLT) 3.4.0 Apr  9 2018 / 21:32:39 

System Uptime: 49 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=240MHz, Vref=3.31V, Core temp=44degC, GYRO=ICM20602, ACC=ICM20602
Stack size: 2048, Stack address: 0x1000fff0
I2C Errors: 1, config size: 2293, max available config: 16384
CPU:32%, cycle time: 35, GYRO rate: 28571, RX rate: 49, System rate: 9
Arming disable flags: RXLOSS CLI MSP

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       1       0    0.5%    0.0%         0
01 - (       GYRO/PID)  31250      29      22   91.1%   69.2%     48908
02 - (            ACC)   1000      10       5    1.5%    1.0%       379
03 - (       ATTITUDE)     99       7       5    0.5%    0.5%        37
04 - (             RX)     49      28      22    0.6%    0.6%        82
05 - (         SERIAL)     99     518       1    5.6%    0.5%        11
08 - (BATTERY_CURRENT)     49       2       1    0.5%    0.5%         1
09 - ( BATTERY_ALERTS)      4       2       1    0.5%    0.5%         0
10 - (         BEEPER)     99       4       1    0.5%    0.5%         4
18 - (            CMS)     59       2       1    0.5%    0.5%         2
19 - (        VTXCTRL)      4       1       0    0.5%    0.0%         0
20 - (        CAMCTRL)      4       1       0    0.5%    0.0%         0
22 - (    ADCINTERNAL)      0       2       1    0.5%    0.5%         0
23 - (       PINIOBOX)     19       2       0    0.5%    0.0%         0
RX Check Function                   3       2                         4
Total (excluding SERIAL)                        98.2%   73.8%
`

Ozzy33, 2018-04-11:
Uart1.

mikeller, 2018-04-11:
Weird. I can't see anything wrong with your config then.

Ozzy33, 2018-04-11:
Like I mentioned, I tried going back to older versions and no problem, it seems to break at BF3.3.1
3.2.5 = OK
3.3.0 = OK
3.3.1 = No Good
3.4.0-755 = No Good

mikeller, 2018-04-12:
Can you please provide the `diff` for a working configuration of 3.3.0?

Ozzy33, 2018-04-12:
`# version
# Betaflight / REVOLT (RVLT) 3.3.0 Mar  2 2018 / 03:59:09 (177472b4f) MSP API: 1.37

# status
System Uptime: 234 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=240MHz, Vref=3.31V, Core temp=55degC, GYRO=ICM20602, ACC=ICM20602
Stack size: 2048, Stack address: 0x1000fff0
I2C Errors: 1, config size: 2252, max available config: 16384
CPU:33%, cycle time: 32, GYRO rate: 31250, RX rate: 49, System rate: 9
Arming disable flags: CLI MSP

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       2       0    0.5%    0.0%         1
01 - (       GYRO/PID)  31250      32      23  100.5%   72.3%    170081
02 - (            ACC)    998   11619       6 1160.0%    1.0%      1540
03 - (       ATTITUDE)     99      11       7    0.6%    0.5%       158
04 - (             RX)     49      26      19    0.6%    0.5%       245
05 - (         SERIAL)     99   11981       1  119.1%    0.5%       181
08 - (BATTERY_CURRENT)     49       2       0    0.5%    0.0%         4
09 - ( BATTERY_ALERTS)      4       2       1    0.5%    0.5%         1
10 - (         BEEPER)     99       5       1    0.5%    0.5%        16
21 - (            CMS)     60       5       0    0.5%    0.0%         7
22 - (        VTXCTRL)      4       1       0    0.5%    0.0%         0
23 - (        CAMCTRL)      5       1       0    0.5%    0.0%         0
25 - (    ADCINTERNAL)      0       3       2    0.5%    0.5%         0
26 - (       PINIOBOX)     19       2       1    0.5%    0.5%         2
RX Check Function                   6       1                        14
Total (excluding SERIAL)                      1266.2%   76.3%

# diff

# version
# Betaflight / REVOLT (RVLT) 3.3.0 Mar  2 2018 / 03:59:09 (177472b4f) MSP API: 1.37

# name
name Raptor4

# resources

# mixer

# servo

# servo mix

# feature
feature AIRMODE

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1600 0
aux 1 26 0 900 1200 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_overflow_detect = YAW
set gyro_use_32khz = ON
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1015
set rc_interp = MANUAL
set rc_interp_int = 14
set serialrx_provider = SBUS
set blackbox_device = SPIFLASH
set min_throttle = 1055
set motor_pwm_protocol = MULTISHOT
set small_angle = 180
set pid_process_denom = 1
set debug_mode = GYRO_RAW
set cpu_overclock = 240MHZ

# profile
profile 0

set setpoint_relax_ratio = 30
set dterm_setpoint_weight = 200

# rateprofile
rateprofile 0

set roll_rc_rate = 250
set pitch_rc_rate = 250
set yaw_rc_rate = 229
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 10
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0`

Ozzy33, 2018-04-12:
Would it help if I load some different Jenkins builds to figure out exactly where it broke, but I would need some pointers where to start (what jenkins build number). Just looking at jenkins, I don't think it keeps the older files (between 3.3.0 and 3.3.1). Let me know if I can help in any way, thanks!

Ozzy33, 2018-04-12:
Good guess!
build 726 broke sBUS, build 725 sBUS is OK!

Not that it matters, why does the working build number 725 is named:
betaflight_3.4.0_REVOLT-725-45e64edb0.hex, indicating its a version of "3.4.0", but it broke in version 3.3.1, not 3.4.0? This had me confused.

AndersHoglund, 2018-04-12:
3.3.1 and 3.4.0 are running in parallel, not sequentially.

fujin, 2018-04-18:
The only thing I've come up with is the rx1/tx1 bridge. :(

Ozzy33, 2018-04-18:
To a pilot, what does that mean?

There is a hex for revolt and revoltOSD, I would guess the regular revolt should not be affected after the OSD changes were made, just a guess.

fujin, 2018-04-18:
It means if it is not working, and the pilot wants it to work, they can bridge rx1 and tx1 pads together (on the bottom of the board - for revolt v1/v2/v3), per our wiki instructions. After doing so, `serialrx_halfduplex` will have no effect - it will work in either `ON` or `OFF`, as the UART1 is bridged together externally instead of internally by `BIDIR` mode. This will not work for RevoltOSD as the pads are not there.

Revolt and RevoltOSD use the same target (it auto-detects the OSD being available).

We haven't figured out why the bidirectionality (internal connection) for tx1/rx1 isn't working in this case. The current theory seems to be that the inverter has to have a floating output when not in use for the RX1/TX1 BIDIR mode to work.

Ozzy33, 2018-04-19:
Thanks for your info.

I see there are three hex's and looks like the OSD is separate hex over the normal (non-OSD) hex.
--
  | betaflight_3.4.0_REVOLT-770-4a5e79a53.hex | 828.69 KB | view
  | betaflight_3.4.0_REVOLTOSD-770-4a5e79a53.hex | 843.84 KB | view
  | betaflight_3.4.0_REVONANO-770-4a5e79a53.hex | 761.33 KB | view

Ozzy33, 2018-05-19:
Is a fix still being looked at or is it a hardware jumper fix only now? tia