betaflight/betaflight #5671: Allow capturing of a short description for PID andRate profiles

Digilus2, 2018-04-11:
This is a feature request 
 _I've searched the open issues and requests and didn't find anything similar._

**Request:**   The ability to have different rate or PID profiles is very useful, however to keep track of which ones you have what settings loaded into it would be useful to allow the user to capture a short description along with the profile.

**Use Case**: The primary use case would be for when switching profiles (rate or PID) via the OSD. 
You can then display the description in the OSD .  So instead of changing to rate profile 3, and PID profile 2 -  you'd be able to see it is your "Race" rate profile and your "Wind" PID profile.
While it is nice to have 9 different combinations of settings - giving them descriptions would probably allow the average user to experiment more with using multiple settings. 

**Limitations**: There would obviously need to be limitations on the amount of characters allowed (as the OSD only has so much space) - 8 characters should suffice (16 would be great but realistically there is no space for it when changing settings via the OSD).

Areas this would impact would be the Configurator, the OSD (both the configuration side and perhaps as an in-flight on screen option) and of course the profile storage array.

pkruger, 2019-03-02:
PR: https://github.com/betaflight/betaflight/pull/7696

pkruger, 2019-03-09:
Hope this PR can still make it into BF 4.0.0.

mikeller, 2019-03-09:
@pkruger: It was submitted after the start of the feature freeze for 4.0, and will go into 4.1, to be merged into `master` right after 4.0.0 has been released.

PR #7696 (2019-05-07): Add PID and Rate profile names, osd elements and cli config
Fixes: #5671
closed 2019-05-07 by mikeller