betaflight/betaflight #5672: Issue with MATEKF411RX 3.3.1 target, will not arm with turtle mode on modes page, and D8 works but not D16

pizzy00, 2018-04-12:
The issue with MATEKF411RX 3.3.1 target, will not arm with turtle mode on modes page, and D8 works but not D16

I am running the latest firmware for my 4-in-1 esc wraith 32bit

mzakk, 2018-04-12:
@pizzy00 I have the same problem. 
So if you delete turtle mode from modes you are able to arm ?

pizzy00, 2018-04-12:
Yes I can arm soon as I remove it and save and unplug usb

richard-scott, 2018-04-12:
You know you now can't arm while connected to the Configurator on USB??
You need to slide the slider in the motors tab before it will allow it.

pizzy00, 2018-04-12:
@richard-scott that is a change made, it is stated the latest release notes =). Anyway, my issue is related to the turtle mode specifically, I only stated that acknowledge I read the notes in a weird way lol.

mzakk, 2018-04-12:
@pizzy00 You got failsafe when turtle mode is on modes?

pizzy00, 2018-04-12:
@mzakk no where did I say that?

turboed13b, 2018-04-13:
It's being worked on.

brycedjohnson, 2018-04-20:
@pizzy00 I have a partial fix for the issues on a similiar crazybeef3FR board.  Some of the D16 issues still remain. Check out the hex on the bottom of #5576.  Try with props off first to make sure everything is working.

closed 2018-05-27 by stale[bot]