betaflight/betaflight #5677: Beep on BB Erase complete - feature request

UltiBots, 2018-04-13:
Please add Beep on BB Erase complete - feature request

mikeller, 2018-04-14:
It actually does beep, the `* * * *` you see are just the visual indicator for beeping, in case you are out of earshot of the craft.

mikeller, 2018-04-14:
No it doesn't. The Dshot ESC beacon signal is designed to be an emergency alert in case of a lost craft, and it's beep strength is set to the same volume as the 'inactivity beacon' on the ESC. For most users, this will probably be configured to a volume that will cause motor damage if it's used over a longer period of time, in order to give maximum sound volume and enable them to find a lost craft. Using this for normal signaling is not a good idea.

Also, since this alert can be activated while disarmed or armed, using the ESC beacon for it will either not work consistently, or else disrupt flight if activated while armed.

UltiBots, 2018-04-15:
All good points. Guess I have to solder on a buzzer :(

closed 2018-04-15 by UltiBots