betaflight/betaflight #5679: Beeper on switch not working Kiss FC

fletch3d, 2018-04-14:
Beeper on switched worked in 3.2.  3.3.1 updated and not working.  Yes, RX_Set is on.  6 Quads, 5 with KISS FC and 1 with OMNIF4SD, that one works fine.  First flashed with defaults and the beepers seem to work.  As soon as I start config, or pull from my dump to update in CLI, the beeper stops working.

[Killer 210.txt](https://github.com/betaflight/betaflight/files/1910573/Killer.210.txt)
[Fletch B1.txt](https://github.com/betaflight/betaflight/files/1910574/Fletch.B1.txt)

mikeller, 2018-04-14:
Updating with a `dump` is not recommended, as this will screw with any defaults that have been changed. Try backing up with a `diff all`, and restoring from that, and see if that fixes it.

fletch3d, 2018-04-14:
Thank you, I will try it.  Have I already messed up or can I do a Diff all and reset? Or do I need to start from scratch with setup?

mikeller, 2018-04-15:
Re-flashing the old firmware, and then restoring the `dump` (which is fine for a restore of the same firmware version), then taking a `diff all`, upgrading again and restoring from the `diff all` should do the trick.

fletch3d, 2018-04-15:
Update on Kiss FC with Betaflight 3.3.1 - For some reason the culprit to the whole switch ordeal for all 6 of my Kiss FC's was to turn on USB - Beep while on bench.  That solved the issue, No the RX_Set.

closed 2018-04-15 by fletch3d

mikeller, 2018-04-15:
Yes, the ON_USB option overrides all others (except for the startup beeps).