betaflight/betaflight #5680: add barometer debug

pitts-mo, 2018-04-14:
Please add barometer debugging.

I suspect any raw barometer sensor data (if) available for temperature, pressure, and altitude, along with betaflight raw and estimated altitude values are desirable for this. I am unsure what is practical with existing sensors and code.

Please see issue #5311 for my motivation and an example where barometer debug is desirable.

mikeller, 2018-04-14:
Have you tried `set debug_mode = altitude`?

pitts-mo, 2018-04-16:
not yet :-) Thank you, -p

pitts-mo, 2018-04-30:
Just a quick update. I will try to come back to this again when I have time. -p

use of set debug_mode = altitude was helpful and educational.
-First value is expected vertical acceleration that I suspect is derived from accelerometer data.
-Second value is calculated vertical velocity that does report drift that corresponds with drift seen in normal OSD altitude.
-Third value is altitude meters x100 that also corresponds with drift seen in normal OSD altitude.

additional notes: 
-The debug OSD data appeared to be slightly delayed when compared with normal OSD altitude value.
-The normal OSD altitude incorrectly displays values between -1 and 0 meters as positive. I suspect this is caused on value conversion for display.

closed 2018-06-06 by stale[bot]