betaflight/betaflight #5681: add manual barometer calibration

pitts-mo, 2018-04-14:
Please add manual initiated barometer calibration.

User barometer calibration initiation methods in order I feel most helpful.
-mode switch
-stick command
-osd option

Please see issue #5311 for my motivation and an example where barometer calibration is desirable.

neilbalch, 2018-05-04:
@pitts-mo Manually initiated calibration or manual calibration? 

_i.e. Are you asking to be able to put in a specific offset or scale or are you just asking for the ability to trigger a barometer calibration from outside the configurator?_

pitts-mo, 2018-05-06:
The ability to manually trigger an instance of barometer calibration.

Currently thinking Baro-Cal switch ON & disarmed state could be used to quickly trigger a single instance only when desired. 

The idea for this issue was related to my comment on #5311
"Today I flew 3.4.0 (2018-04-09) with performBaroCalibrationCycle(); tucked into disarm(void) and find the behavior helpful when I fly for a good 30 seconds and land before continuing... For my case I still think debug would be necessary to find root cause of altitude deviations."

neilbalch, 2018-05-07:
@mikeller beat me to it, #5834 addresses this problem by resetting the barometer altitude to 0 when arming.

mikeller, 2018-05-07:
@neilbalch: Wasn't me, it was mostly @dbasch, with me just fixing it up to work without a GPS present. It's all part of the move to add a 'return to home' facility to Betaflight.

pitts-mo, 2018-05-07:
5834 does not offer feature equivalent to manually initiated barometer re-calibrate.

I do estimate desire for this feature from average betaflight user is pretty low... Myself I wanted to put priority on baro debug #5680 but I have low skill level, time, and still need to find/learn a real code editor for this in ubuntu.

mikeller, 2018-05-07:
@pitts-mo: As I stated above, the new code that deals with the altitude estimate is part of a move to implement a basic 'return to home' facility for Betaflight, and #5834 is just extending this to make it useable without GPS. For return to home, what is needed is the relative altitude compared to the point of take off, and not to a user specified reference.
Unless there is a good use case for manual setting of the altitude reference, I would prefer to keep automatic setting of 0 altitude on arm, in order for this not to interfere with the return to home functionality.

P.S.: Vim works just fine for C code, and it's often a good idea to start out on an unfamiliar project by using a plain text editor, in order to get yourself acquainted with the code base. The command line build system works pretty much out of the box in ubuntu.

closed 2026-09-26 by blckmn