betaflight/betaflight #5682: add cli setting baro_cal_on_first_arm

pitts-mo, 2018-04-14:
Please add cli setting baro_cal_on_first_arm to perform barometer calibration on first arm with functional implementation similar to the gyro_cal_on_first_arm feature.

Note: This feature seems desirable in general but I do not find it helpful for the altitude inconsistency I currently see in my build. Please see issue #5311 for some background and an example where barometer calibration was speculated to be desirable.

mikeller, 2018-05-25:
@pitts-mo: This has now been implemented in an unconditional way as part of GPS rescue in 3.4 (better make it establish correct ground altitude **every time** before take off, in case it's going to be used for an emergency landing).

Can you please check, confirm it's working for you, and close this issue?

pitts-mo, 2018-05-27:
@mikeller 
betaflight_3.4.0_KAKUTEF4V2-20180527_153700 (git clone time) 
Yes, The baro/altitude is re-zeroed as you describe this seems ideal for most users and does appear to satisfy the primary intent of this issue. 

However I would like to note this behavior is exacerbating the baro inaccuracy I see on my build since it is re-zeroing several meters of negative drift accumulations prior to arm then adding that to the already excessive positive drift I see within the first 20-30 seconds of flight. I really just need to look at adding baro debug to resolve the cause of drift since my trick of re-calibrate on every disarm is hardly ideal either.

closed 2018-05-27 by pitts-mo

mikeller, 2018-05-27:
@pitts-mo: I think setting the altitude to 0 on arm is a step in the right direction (and it's needed for GPS rescue anyway). As a side effect it gets rid of drift that occurred before arming.

Drift after arming is a different problem, and cannot be addressed in this way. The original code tried to use ACC data (if present) to reduce the influence of BARO drift. But it did not work particularly well, and it was messy. Maybe we should look into bringing this back in a clean form.

rgp1942, 2018-07-21:
So if I understand what you are saying here, is that in BF 3.4, by default the barometer is reset to ground level (0) automatically, and no pilot action or BF programming/configuration is needed in order to use the barometer + OSD to get an on-screen altimeter.

mikeller, 2018-07-21:
@rgp1942: Yes, the altitude tracking is set to 0 on arming - this is what is used in GPS rescue to land the craft.