betaflight/betaflight #5687: BF3.3.2 / BF3.4.0 - same target but different defaults on esc / motor protocol

TheIsotopes, 2018-04-15:
since `build #42` the target MATEKF405 brings different defaults on MatekF405-OSD and MatekF405-Mini ... the osd is going to brushed esc / motors support as defaults but the mini not ... on `build #41` both have the same oneshot125 support as default ... same issue also on bf3.4.0 targets ... imho on both the defaults must be oneshot125

@etracer65 pls can you check this and fixed ... imho i think that's wrong and not wanted

TheIsotopes, 2018-04-15:
I'm understanding now whats wrong ... imho it's my hardware configuration ... using the f405-osd in an tricopter and on motor 1 pin i have the servo ... fc only makes what he does, it finding not an esc on motor 1 pin and he think it's an brushed motor ... ok

@mikeller  If it so, then we can close here

mikeller, 2018-04-15:
@TheIsotopes: Yes, that would do it. The autodetection is pretty basic, and only checks the level at the motor 1 pin (i.e. the default pin after a reset) and uses it to determine if it's connected to an ESC or a FET. The main reason for this is to avoid the situation where motors start to spin on brushed boards after a reset to defaults, because the OneShot 125 idle signal acts as non-idle PWM signal. The reverse (motor PWM signal going to ESCs) isn't a problem, since motor PWM will not be recognised as a valid input signal by the ESCs, keeping them in an unarmed state.

TheIsotopes, 2018-04-15:
thanks for the explanation ... since this is not a bug i will close here

closed 2018-04-15 by TheIsotopes