betaflight/betaflight #5691: BF 3.3.0/OMNIBUS F3/ rssi on OSD is stuck

Bugg69s, 2018-04-15:
I have flysky i6 running FlyPlus 1.7.5 and I have setup the error (rssi) to aux. 
In receiver tab its Ok.
But on osd it is stuck at 40.

I have tried to set rssi to a potentiometer on transmitter. When I rotate it on receiver tab it is moving but in OSD the value isn't changing. 
( when I'm rotate it over like 3/4 buzzer starts beeping)
Idk what can be the problem

Bugg69s, 2018-04-15:
*no it isn't beeping I had the same potentiometer set to beeper😅

etracer65, 2018-04-15:
Have you set the rssi_channel correctly?

```
# get rssi_channel
rssi_channel = 0
Allowed range: 0 - 18
```

Bugg69s, 2018-04-15:
Idk, in receiver page it works on 100% right.

etracer65, 2018-04-15:
The receiver tab just shows what the channels are doing.  You need to set `rssi_channel` to the channel you're using for RSSI (the channel that shows the activity in the receiver tab).  Otherwise the RSSI logic doesn't know the source channel to use.

Bugg69s, 2018-04-16:
I have rssi setup to aux 6 and in top right I have selected rssi to be on aux 6

Arakon, 2018-04-16:
Make sure you don't have rssi_adc active at the same time.

Bugg69s, 2018-04-16:
Thanks, that was it 😄😄😄😄😄🐙

closed 2018-04-16 by Bugg69s