betaflight/betaflight #5692: BF 3.3.1 / IBUS: rssi channel not active after commit #5644

AirBreak69, 2018-04-15:
#5644 introduced some config checks regarding rssi settings.
Beginning with build 2171 on my DYSF4PRO using IBUS the rssi channel is reset to zero after reboot.
Reverting to build 2170 brings the rssi channel back.
Looking at the code changes you can find that the rssi channel is reset if  rssi_src_frame_errors is ON.
Setting rssi_src_frame_errors = OFF re-enables the rssi channel but in the OSD the rssi value remains zero.

A relevant part of my settings:
# get rssi
rssi_channel = 6
Allowed range: 0 - 18

rssi_src_frame_errors = ON
Allowed values: OFF, ON

rssi_scale = 30
Allowed range: 1 - 255

rssi_invert = OFF
Allowed values: OFF, ON

mikeller, 2018-04-15:
Thanks. The priorities of the different RSSI inputs need looking at.

AirBreak69, 2018-04-19:
Can confirm: fixed with #5703

closed 2018-04-19 by AirBreak69