betaflight/betaflight #5698: Motor issue start to twitch  Omnibus F4 - DSHOT

trunglp, 2018-04-16:
# Last week i build mini quadcopter and i problem with my motor . It start same twitch  

1. Harware
- [Min F4 Flytower (Omnibus F4) ](https://vi.aliexpress.com/item/Mini-80mm-MM80-Racing-frame-Quad-Mini-F3-F4-Flytower-Flight-Controller-Integrated-OSD-4-in/32804846362.html?spm=a2g14.10010108.1000016.1.2a1f18b7F8uNOO&isOrigTitle=true)
- "ESC 4 in 1" Blheli 15A
-  Brushless Motor 1104 7500KV :
-  [Flysky i6 and ibus ](https://des.gbtcdn.com/uploads/pdm-desc-pic/Electronic/image/2018/03/01/20180301163429_28214.jpg)

2. Software
- I use Betaflight 3.3.1 and betaflight configurator 10.2 
- I use IBus connect UART1 (Pin Rx1 connect to iBUS ) and it working 
- I set protocol : DSHOT300 
- I use (Gyro updata frequency / PID loop frequency) =>  8khz/2kHz (default in betaflight)
- I use Dynamic_filter 

And this is problem

- When I go to the 'Motors' tab and enable the 'Motor test mode' switch (REMOVE ALL PROPELLERS FIRST); => every motor is smooth. [View image](https://user-images.githubusercontent.com/38026441/38794156-51f3960c-417e-11e8-8efc-69a120e8c601.jpg)
- I use BlHeliSuite test => every motor is smooth. and config in BlHeli is good  [view image](https://user-images.githubusercontent.com/38026441/38796756-8b57c50a-4186-11e8-99bb-4fc5440bb137.png)

**But I using control from the radio, every motor is not smooth. Every motor start not the same, motor start to twitch.**   [View error ](https://user-images.githubusercontent.com/38026441/38794155-51beffaa-417e-11e8-9d8c-1e69bf8934e7.jpg)
I fly to test => my quad is crash
I change any DSHOT protocol but not working 
 

This is log  [BTFL_BLACKBOX_LOG_20180416_141719.zip](https://github.com/betaflight/betaflight/files/1914288/BTFL_BLACKBOX_LOG_20180416_141719.zip)

how can i troubleshoot my quadcopter Can you help me.

[image]

[image]

[image]

[image]

```
# diff
# version
# Betaflight / OMNIBUSF4 (OBF4) 3.3.1 Apr  3 2018 / 09:54:44 (4b7eaf79d) MSP API: 1.37
# name
# resources
# mixer
# servo
# servo mix
# feature
feature -OSD
feature MOTOR_STOP
feature DYNAMIC_FILTER
# beeper
# map
# serial
serial 0 64 115200 57600 0 115200
# led
# color
# mode_color
# aux
aux 0 0 1 1700 2100 0
aux 1 1 2 900 1700 0
# adjrange
# rxrange
# vtx
# rxfail
# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set serialrx_provider = IBUS
set motor_pwm_protocol = DSHOT300
set align_board_yaw = -90
set beeper_dshot_beacon_tone = 5

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
# rateprofile
'''
``

etracer65, 2018-04-16:
Try increasing your "Motor throttle idle value" on the configuration tab (right below where you select DSHOT300).  The default value of 4.5 is probably too low for those little motors.  You'll probably need a value of around 6.0 or maybe more.  You could use the motors tab to find the minimum value.  Individually spin each motor (without props!) and try to find the lowest setting where the motor spins reliably.  Use your finger on the side of the motor to introduce a little drag.  Do this for each motor and find the highest required value.  Then convert this into a percentage to use as the idle value.  For example, 1060 would be 6.0%; 1055 would be 5.5%, etc.  Probably a good idea to add a little to your experimentally determined value to add a safety margin.

trunglp, 2018-04-16:
@etracer65 : Thank you for reply.
I try increasing "Motor throttle idle value" to 6.0 but not working => All motor is twitch. 
I check motor tab and i see the minimum value is 1020. ( i see blheli ppm min throttle: 1000 us and max ppm 2000us)  [view image](https://user-images.githubusercontent.com/38026441/38796756-8b57c50a-4186-11e8-99bb-4fc5440bb137.png)

This is the video after i change  "Motor throttle idle value" => 6.0

["https://youtu.be/UwBYBNSKGQA"](https://youtu.be/UwBYBNSKGQA)

etracer65, 2018-04-16:
Not sure I see any problem in the video you posted - definitely no "motor twitching".  Are you concerned that the motors run at different speeds and are speeding up when armed?  If so, that's perfectly normal.  You probably have the flight controller in a level mode (angle or horizon) and the flight controller is trying to correct the orientation.  But it can't because there are no props to allow the quad to move.  So it tries harder and that's why the motors increase (called I-term windup).  Disconnect from the configurator, put the props on and fly - should be okay.

etracer65, 2018-04-17:
There are lots of tuning resources available, but generally hot motors means a noisy system with too little filtering.

Use the rcgroups forums for tuning related help:

https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

You can also refer to tuning/setup guides like:

https://www.youtube.com/watch?v=8vJCrHj9s6s
https://www.youtube.com/watch?v=XmQ28MuAFf0

I'm going to close this issue as there is no problem with the Betaflight firmware.

closed 2018-04-17 by etracer65

scuenca, 2018-10-08:
hi, 
I have the same issue. I have esc that supports dshot and i have configured them DSHOT600.
In, betaflight, with the motors test option, the motors are running ok, with the same power all of them. But using control from the radio, every motor is not smooth. Every motor start not the same, motor start to turn with different powers. Here is the image of the betaflight graphic when i use throttle:
[image]

etracer65, 2018-10-08:
@scuenca This is completely normal. The PID controller is trying to "fly" the quad so the motor outputs are different based on the corrections needed for flight. With no props the flight controller can't correct so it keeps trying to adjust the flight - resulting in different motor outputs.

Put the props on and test in a safe area (not connected to the configurator).

scuenca, 2018-10-08:
Ok, i understand. I didn’t know it.
Thanks!