betaflight/betaflight #5710: No Yaw in velocidrone via Flight Controller Wireless Sim

rhys100, 2018-04-18:
BF is not senind yaw the USB joystick interface, fix is currently to sassign yaw to aux1 in your model memory on your transmitter.

mikeller, 2018-04-18:
Quick test with `jstest` shows correct values for all channels. Looks like this is a problem with velocidrone. Closing.

closed 2018-04-18 by mikeller

ashleybadavis, 2018-04-23:
We have had two customers report this problem to us now (velocidrone) and in both cases the yaw channel does not show up in the windows joystick controller created by betaflight. With that in mind (and I haven't tried this myself) I would appreciate it if you looked at this again as this is not in my opinion anything to do with velocidrone. We read all joystick input directly from the windows joystick controller - if there is no input there then we cannot assign it inside the sim.

DieHertz, 2018-04-23:
I suggest you to look at it since you have experience with your simulator, and betaflight's code is open for you to peek at.

mikeller, 2018-04-23:
@ashleybadavis: As stated above, testing with `jstest` under Linux shows all 4 channels moving with stick input as they should. I'd suggest you put some scrutiny into how the device is recognised by windows, and if the correct driver is used.

ashleybadavis, 2018-04-23:
I think this is where the problem lies - just assigning channels whilst fine in Linux is not fine in windows. In windows the different analogue axis have to be assigned to one of eight potential anologue axis - x axis, y axis, z axis, x rotation, y rotation, z rotation and the two slider axis. If you do not assign these correctly then the axis will just not be seen by any program using the windows joystick interface - which is what velocidrone uses. The fix for our customers has been to use a channel that is assigned to one of the eight listed above - aux 1 for example and then everything works as it should.

ashleybadavis, 2018-04-23:
We will take a look at the code and see if we can see what needs to change for it to work on windows - I'll let you know

mikeller, 2018-04-23:
Go for it. The way we are sending the axes over HID is pretty basic:

https://github.com/betaflight/betaflight/blob/bede91212866f7068e7a8da3f5e9effe4d78080d/src/main/fc/fc_tasks.c#L155-L168

Maybe this could be addressed by a custom driver?

ryamoo, 2018-06-11:
Was this resolved in RC1 or is yaw channel still missing on Windows?

ashleybadavis, 2018-06-12:
I'm afraid I haven't had opportunity to look at this, it is on our development schedule but we haven't been able to look at it yet. The issue is as far as we know is a problem of not assigning channels correctly in the windows USB joystick interface as there are only eight defined axis to which you can send analog data and for whatever reason yaw is not being assigned to one of these (this is our guess as this is a difference  how Linux / OSX work).

mikeller, 2018-06-13:
Would somebody mind testing #6099 with Velocidrone? Let me know what target you need if you can't build it yourself.

Redeven2, 2018-06-21:
Same problem here

DieHertz, 2018-06-21:
Did you try latest dev build from Jenkins or GitHub?

ryamoo, 2018-06-21:
@DieHertz  I'm willing to give it a try - I've had a look through how to build myself. Is it just a case of downloading the main git, or do I need to specify a certain branch to include the #6099 fix?

DieHertz, 2018-06-21:
You should just download a HEX from https://ci.betaflight.tech

ryamoo, 2018-06-21:
@DieHertz thanks - I didn't realise that existed :-D I'll try it tonight.

mikeller, 2018-06-21:
@ryamoo: 3.4.0-RC2 (available in the configurator) has the fix as well, please just try that one.

Redeven2, 2018-06-21:
I believe that is what I am running will check and see. Thank you

red-barran, 2018-06-23:
Hi, please do let me know if this is the correct place to comment or not. 
I'm seeing a similar issue but with FPVFreerider. 
Using Matek F405 CTR, Betaflight 3.4 RC2. In BF Configurator the 4 main channels (Pitch/roll on the right, yaw, throttle on the left - mode 2) display correctly in the receiver tab.
In the Matek Joystick device in Win 10 Devices, MatekF4 game controller settings, the behaviour seems to be correct, right stick moves the dot in the square, left stick changes the X rotation and Y rotation sliders, see capture below.
[image]
In FPVFreeRider, the right stick seems to behave, but left stick gives crazy response in the Calibrate joystick section

Azuxul, 2018-08-19:
Same problem as @red-barran in BF 3.5.0.

dkustec, 2020-08-30:
This PR #10150 could solve this problem in Windows, jfyi.
You can test those compiled hexes there.

dkustec, 2020-08-30:
The #10150 could #10150 possibly can solve this on Windows.

dkustec, 2020-08-30:
The #10150 could possibly solve this on Windows.