betaflight/betaflight #5713: Performance profiles

rustbg, 2018-04-19:
Defaults are far from adequate for a good setup nowadays. Hence deliver mediocre performance for people that have no idea or interest to fiddle with filters, sliders and so. So they never have the possibility to experience the best ot betaflight. 
Would be great to have a few preset performance profiles (similar to iNav) that don't have as much filtering, less D gains and relax ratio to 0 for example (one of the best thing happened in betaflight in years, thanks to ctzsnooze!) 

Cheers guys! You are the best!

digitalentity, 2018-04-19:
Possibly also less P-gains. Defaults are too high for modern powerful high-kv motors with kv=2600-2700 and modern high-power ESCs and batteries.

theSkonk, 2018-05-04:
And filter profiles with the new filters. If you don't turn them on your not getting the magic so a couple of different levels of pid and filter would be good for people who don't understand tuning

mikeller, 2018-06-03:
Not sure that it's worthwhile spending valuable flash space on implementing preset profiles in the firmware, especially since we already have to shave the feature list down to the bare minimum on F3.
This should probably rather be done in the configurator.

theSkonk, 2018-06-03:
How about a new base profile that's a bit more refined for modern qwads. It's the only reason 'butter flight' flys well out of the box. It has basic settings that work better for most people

mikeller, 2018-06-03:
@theSkonk: There is work under progress to define new default settings in #6036, feel free to join in to the discussion.

closed 2018-07-11 by stale[bot]