betaflight/betaflight #5719: inav 1.8 throttle set less than 1100, motors jump to 1600+

mss2015, 2018-04-19:
SETUP: I'm using a sirius airhero3 pre-flashed with inav 1.8 and configured for quad-x. Quad is a FPV Flip frame with 9 inch props, 4S batter, 1100 kv NTM motors, and Afro 30amp ESCs. the ESCs and motors calibrate perfectly in Configurator. 
ISSUE: In Angle or Horizon mode, when I arm and raise throttle, throttle jumps immediately to 1600+. I have to hold the quad down or in my hand to avoid it skyrocketing or crashing. I raise/lower throttle several times and turn transmitter on/off, and after 30 sec to 1 min, throttle level stabilizes to the ~1100 it's set for minimum, and quad flys great and the throttle jump issue goes away. I've read various posts about airmode but have not toggled that yet. My issue is nearly identical to what's described in the post below. Solution there was to reflash the board, which is my next step. Thought I'd ask first if others have seen this behavior.

https://github.com/betaflight/betaflight/issues/281

Any feedback is appreciated!

jflyper, 2018-04-19:
You might want to re-file this ticket at https://github.com/iNavFlight/inav/issues

closed 2018-04-19 by jflyper

mss2015, 2018-04-19:
Thanks for the response and direction - I've re-filed where indicated