betaflight/betaflight #5725: MCU Core temperature warning/critical threshholds

fujin, 2018-04-19:
I have seen in a couple of cases now where the MCU core temp. reaches critical levels, eventually resulting in a FC thermal shutdown. This typically presents as an OSD, telemetry, RC freeze, with a solid status led indicator. It can be triggered without overclock in cases where a vtx running hot is mounted on, or close to, the FC stack. I have triggered it also via overclock at 240mHz (but generally running 192/216 avoids.)

In order to keep a closer eye on it, I now have MCU core temp on OSD.

In addition to the OSD display, I was thinking about adding configurable warning/critical threshholds, with sensible defaults. e.g.

mcu core temp >=60C -> warning
mcu core temp >=80C -> critical

Thoughts? @jflyper perhaps..?

mikeller, 2018-04-20:
Good point. If this is starting to become a problem we should add warnings.

jflyper, 2018-04-20:
I think it's a very good idea; will rule out a case from _"drop from the sky like a stone"_ type incident 😀 

We might even think of adding core temperature of, say, under 75 degC be required to arm (user adjustable?)

conkerkh, 2018-04-30:
Drops like a stone or goes full throttle?

wind0r, 2018-04-30:
I think that a locked CPU will not be able to send valid data tothe motor so the quad should drop like a stone.

rodizio1, 2018-05-03:
Not the CPU, but the hardware timer stuff may continue running and be 'frozen' at the last values the CPU has written to it. PWM to the ESCs and servos is probably such a thing, when the CPU crashes, the PWM values may stay the same causing the aircraft to continue with those servo/throttle positions until the battery runs empty. Classic flyaway.

conkerkh, 2018-05-03:
@rodizio1 I thought about this scenario but I think it affects only PWM signal, in DShot it should be fine as DMA doesn't run in Circular mode (I think)

brucesdad13, 2018-05-22:
I've had this happen a few times with a whoop where the VTX was mounted under a vacuum formed canopy on top of the overclocked F4. Usually it drops like a stone but once it went full throttle and was unresponsive to stick input. Can the clocking be changed on-the-fly? If so, perhaps dropping to 216MHz if the temp crosses a certain threshold?

brucesdad13, 2018-05-22:
For whatever reason, 216MHz never seems to go into thermal shutdown

mikeller, 2018-05-22:
Did you ever check how much power / run time you trade off on a whoop by overclocking? ;-)

conkerkh, 2018-05-22:
Considering what he said that whoop went full throttle on thermal shutdown, I would add ADC watchdog in core temp readings and shutdown timers at around 80deg before everything locks up.

PR #5980 (2018-05-27): Added warning for core temperature to OSD.
Fixes #5725.
closed 2018-05-27 by mikeller