betaflight/betaflight #5731: BF 3.4.0 I-Term not working on all axis with latest build

TheIsotopes, 2018-04-20:
using latest `build #777` ... no axis is stable when flying and must be counteracted ... in bblog we can see that all axis have no iterm action

[BTFL_BLACKBOX_LOG_TruX_20180420_200134.TXT](https://github.com/betaflight/betaflight/files/1933266/BTFL_BLACKBOX_LOG_TruX_20180420_200134.TXT)

etracer65, 2018-04-20:
I can replicate - looking into it.

etracer65, 2018-04-20:
It's been narrowed down to #5674 causing some unexpected side effects and it's being worked on.

DieHertz, 2018-04-21:
Be careful when using `FAST_RAM`, for non-trivially-initialized variables you should use `FAST_RAM_INITIALIZED`.
This is what caused Iterm to be inactive unless anti-gravity kicked in.

PR #5733 (2018-04-21): Added FAST_RAM_INITIALIZED
This PR adds `FAST_RAM_INITIALIZER` storage class for use with non-trivially initialized variables (e.g. initialized to anything different from 0).

Please keep the usage scenarios in mind in case you're going to add new `FAST_RAM` variables:
```cpp
FAST_RAM int triviallyInitializedVariable1;
FAST_RAM int triviallyInitializedVarible2 = 0;
FAST_RAM bool triviallyInitializedVariable3 = false;
FAST_RAM void *triviallyInitializedVariable4 = NULL;
FAST_RAM_INITIALIZED int nontriviallyInitializedVariable = 1;
```
After enabling `FAST_RAM` for F4 in #5674 an issue with `ite [... cut]
closed 2018-04-21 by DieHertz