betaflight/betaflight #5732: Reverse Motor dont work

Greenwolff91, 2018-04-20:
Hello,
i want to reverse my motors on a stock immersionrc vortex 150mini with newest betaflight 3.2.2 Firmware, but i doenst takes effect. The display of reversing in betaflight ist correct.

therealjonnyd, 2018-04-20:
Are you changing your motor direction in your ESCs or by swapping motor wires, as well? That's a necessary step.

Greenwolff91, 2018-04-20:
I am new in fpv racing. I buy it from a fried he has a dx7 and i have a dxe. He flash a vortex 180mini on it. i fly with this software correctly, but when arming the propeller donst spin correctly and the osd text didnt goes of. So i flash the actual fc firmware and osd firmware, now arming and osd works correctly but all motors spin in the wrong direction. I want to solve the problem with konfiguration and dont want to change/touch the stock esc and connections

Greenwolff91, 2018-04-20:
I have also the problem that only 2 switches on my spektrum dxe can be configured. The other doesnt take effect, habe you ne idea for the problem

mikeller, 2018-04-21:
@Greenwolff91: Betaflight 3.2.2 is not the latest version, and is missing some safety features. Please update to the latest version (currently 3.3.1).

To change the rotation direction of your motors without soldering, you will have to reprogram your ESCs. Contact the manufacturer to get a configuration tool for the ESCs.

closed 2018-04-21 by mikeller