betaflight/betaflight #5735: Full Yaw causes disarm

gonatienza, 2018-04-21:
Hi,

I'm testing a new build with Betaflight / OMNIBUSF4SD (OBSD) 3.3.1 Apr  3 2018.  Full CLI dump attached.

When I test (props or no props) full yaw (either way) it disarms almost every time.  Most of the times when I'm in air mode.  But it's happened in acro mode, too.  I'm also attaching two blackbox tests.

HW:
-Hobbywing 40A 4in1 ESC
-Hobbywing F4
-Frsky R-XSR Receiver

Any ideas?

Thanks.

[dumpBF.txt](https://github.com/betaflight/betaflight/files/1933952/dumpBF.txt)
[Blackbox.zip](https://github.com/betaflight/betaflight/files/1933953/Blackbox.zip)

jflyper, 2018-04-21:
anti-taz

gonatienza, 2018-04-21:
Just check the documentation around this, and that shouldn't be the case.

From https://github.com/betaflight/betaflight/wiki/Runaway-Takeoff-Prevention

Successful Flight:

Once the craft successfully takes off and is in controlled flight the feature will deactivate for the remainder of the battery. The detection of successful flight involves looking at throttle level and control stick activity along with the craft responding correctly to the PID controller over a period of time. By deactivating the feature after successful flight is detected we prevent subsequent inadvertent disarming caused by crashes or gate/branch clips.

Arming with no props and moving the R/P/Y sticks significantly may trigger a disarm. This is normal and is caused because the PID controller is trying to control the craft and it's not responding properly (because there are no props!). As a result the PID_sum values will grow as the PID controller begins to push harder and harder to make something happen. This can end up looking like a runaway PID_sum event. The feature has provisions in place to allow limited bench testing without disarming. Heavy yaw movements could still trigger a disarm but generally "wiggling" the sticks to see the motor responses won't be a problem. Testing while connected to the configurator will temporarily disable runaway takeoff prevention and allow unlimited bench testing without props.

closed 2018-04-21 by gonatienza

Quick-Flash, 2019-01-01:
I just recently had the same issue on a quadcopter using betaflight 4.0.0 once again disabling this fixed the problem.

fujin, 2019-01-02:
@Quick-Flash no, you explicitly ignored our direction to properly configure runaway takeoff prevention for your craft, including a link to the documentation and suggested parameter to tune.