betaflight/betaflight #5736: Tilt servo not able to use full range of motion.

importon, 2018-04-21:
I've been using the servo gimbal function of betaflight for some time and I love it.  However, if I just connect my servo directly to my receiver channel it will rotate about 180 degrees from one endpoint to the other, but if betaflight is controlling the servo the most range I can get is closer to 100 degrees.  I've maxed out all the values in the configurator.  Is there some hidden way to unlock the full range of my servo in betaflight?

jflyper, 2018-04-21:
Use `CHANNEL_FORWARDING` instead of `SERVO_TILT`.

For further assistance,  please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

closed 2018-04-21 by jflyper

importon, 2018-04-21:
will I still get the gimbal/autolevel  motion sent to the servo if I use channel forwarding?  Is there any documentation anywhere on how to do this?

jflyper, 2018-04-22:
@importon Ah, if you want the gimbal control, you havre to stick to `SERVO_TILT`.

Now, when I test the `SERVO_TILT` feature, I can make the servo outputs swing between MIN and MAX which I configure in servo tab.

What is the PWM value for 180 degree rotation for your servo?