betaflight/betaflight #5744: Motors shut off only on max Yaw/Pitch/Roll (Betaflight F7)

MIKE-GR, 2018-04-22:
Hi all.
I just made a quad with :
- Betaflight 3.3.0 (F7)
- DJI Snail propulsion system
- Fr-XSR
- BN-GPS
- RunCam mini
- Eachine VTX03

After the basic configuration for the FC, aux channels, channel calibratio, telemetry, etc.

I found that after arming the quad and when y try to go to the max Roll/Yaw/Picth the motors shutdown and in the OSD screen the information that normaly appears when I shutoff the quad is showed, so I have to re-arm to enable to quad again.

But this is not happening for the throttle, as I have no problem to go to max power.

About the basic verificacions, the RC is sending the corrects values from 1000 to 2000 for each of the 4 channels and I jave done the calibration  in order to have the exacts values in the RC and in the FC console, the YAW by default is 1500.

I did a test and record in the blackbox the activity (attaching the data file), one thing that I want to know is if the rccommand[yaw] needs to be 1500 instead of 0 when there is no activity, because according to the screen shot (attached ), the rcCommand[throttle] is 1000 that is the same that I configure in the RC section.

About the wiring, I have done a re-make of the dron, first I leave only the motors and the r-xsr connected (wiring again all the connections) and the problem persist, so I don't think that is the  problem with the wiring or vibrations. the other fact is that the throttle works well and with the 4 motors.

The last thing was to update the receiver and radio (inside Taranis QX7), without success.

So this is why I came here at the end, because I was review all the other options that I get from the blogs.

could you help me, Any idea?

[diff_BTFL_backup_20180418_201610.txt](https://github.com/betaflight/betaflight/files/1935448/diff_BTFL_backup_20180418_201610.txt)
[BTFL_BLACKBOX_LOG_20180421_223000.zip](https://github.com/betaflight/betaflight/files/1935450/BTFL_BLACKBOX_LOG_20180421_223000.zip)

DieHertz, 2018-04-22:
Antitaz

MIKE-GR, 2018-04-22:
greate that's it.. many thanks..

closed 2018-04-22 by MIKE-GR

DieHertz, 2018-04-22:
Good
It's a safety feature, of you're going full roll/pitch/yaw without props, I wouldn't recommend doing that. If you're with props it shouldn't occur under normal conditions, and after you've taken off it's disabled and can't occur in flight

MIKE-GR, 2018-04-22:
Yes, I just read the documentation about it, interesting safety feature, I will try now to fly it and let's see what happens. thanks