betaflight/betaflight #5751: BLHeli ESC flash failing with latest BF dev builds

ethomas997, 2018-04-23:
I'm finding that when flashing RacerBee speed controllers with BLHeli via ESC pass-through, using the latest Betaflight development build ([#786](https://ci.betaflight.tech/job/Betaflight/763)), the flash fails midstream, leaving the speed controller non-operational.  I see the same trouble with build #[672 ](https://ci.betaflight.tech/job/Betaflight/672)(2018-03-15), so it looks like the problem goes back at least that far.  When I put the flight controller back to Betaflight v3.3.1, the ESC flashings succeed.

Hardware is SPRacing F3 FC, and Super RacerBee 30A (BLHeli-S J-H-90) speed controllers.  Settings are at defaults, and using BLHeliSuite 16.7.14.9.0.1, flashing BLHeli-S 16.7

BTW, if you run into this and end up with a flashing-failed ESC, you can recovering it by keeping it powered on (important), putting the flight controller back to Betaflight v3.3.1, and repeating the ESC flash in BLHeli (need have only that ESC selected via mouse right click).  If you power down and restart the ESC in the flashing-failed state then it will no longer have a working bootloader and will need to be flashed via the C2D/C2K pins.

--ET

DieHertz, 2018-04-23:
Could you identify the latest build # which doesn't exhibit the issue?

ethomas997, 2018-04-23:
@DieHertz Sorry, no, I didn't have time to narrow it down further.  The most I can say is I verified that BF v3.3.1 and the latest 'v3.3.x-maintenance' branch work OK.

ethomas997, 2018-04-23:
Also, with the latest dev version of Betaflight on the FC I am able to update the configuration on the ESCs (i.e., to reverse motor direction), so the basic read/write-settings communication is alright.  Seems to be just the flashing that fails.  It gets part-way through flash write, and then fails with an error popup (offering a retry, which will also fail).

DieHertz, 2018-04-23:
Do you have a log from either BLHeliSuite or BLHeli Configurator?

ethomas997, 2018-05-23:
I haven't had time to look at this more; I was hoping that others might be able to test and confirm it.  If this issue hits users down the road they will be very unhappy if they go to flash and end up with non-operational speed controllers.