betaflight/betaflight #5759: Spikes in motor traces with rc_interpolation, and amplified by throttle boost

smasslich, 2018-04-24:
[image]

I am seeing large spikes in the motor traces that are 9, 18, and 27 ms apart throughout the log.  This is with rc_interpolation set to manual, and RPYT.  Throttle boost amplifies these spikes.  Results in heat generation in motors.  

[LOG00001 thru 8.zip](https://github.com/betaflight/betaflight/files/1940716/LOG00001.thru.8.zip)

CLRacingF4 FC, DSHOT600, BLHELI-S, XM+ RX, BF 3.4.0 Build 782
For all logs:  gyro_hardware_lpf = EXPERIMENTAL, D8 and same model on Taranis Plus
LOG00001:  rc_interp = MANUAL, rc_interp_ch = RPYT, rc_interp_int = 14, throttle_boost = 50, throttle_boost_cutoff = 40
LOG00002:  changed throttle_boost = 0
LOG00003:  changed rc_interp_ch = RP
LOG00004:  changed back to rc_interp_ch = RPYT, throttle_boost = 50, throttle_boost_cutoff = 40, and updated rc_interp_int = 19
LOG00005:  rc_interp = changed rc_interp_int = 9
LOG00006:  rc_interp = changed rc_interp_int = 27
LOG00007:  rc_interp = changed back to rc_interp_ch = RP, rc_interp_int = 14, throttle_boost = 50, throttle_boost_cutoff = 40
LOG00008:  changed throttle_boost = 0

[Chameleon Diff BF 3.4.0 Build 782.txt](https://github.com/betaflight/betaflight/files/1940717/Chameleon.Diff.BF.3.4.0.Build.782.txt)

I do not see these spikes with the Matek F405-OSD FC and XSR, R-XSR rx, although I do not know if the rx has anything to do with it.  

Thanks!

flint723, 2018-04-24:
I also saw this once.  Firmware is 3.4 but a little older (throttle boost wasn't merged yet).  This is on impulsercf3 target.  RF link is DSMX 11ms, SRXL, SPM4648 receiver.  Similar to smasslich findings, spikes appear either at the RF frame rate, or at a multiple of the frame rate.

Log with throttle interpolation off (no spikes):
[image]
[flight 14,15 5s throttle interp off.zip](https://github.com/betaflight/betaflight/files/1940756/flight.14.15.5s.throttle.interp.off.zip)

Log with throttle interpolation on (rc_interp_interval =11ms) (has spikes):
[image]
[flight 16,17 5s throttle interp on.zip](https://github.com/betaflight/betaflight/files/1940758/flight.16.17.5s.throttle.interp.on.zip)
[helix diff throttle interp on.txt](https://github.com/betaflight/betaflight/files/1940759/helix.diff.throttle.interp.on.txt)

Targets I never did and still don't see these spikes on (with throttle interpolation (interval 11) and boost enabled):
matekf405 (Matek F405 mini)
omnibusf4 (HGLRC 20x20 F4 v2)

apoc, 2018-04-25:
Hi I am trying to analyze the issue, I prepared for You custom HEX files with few changes, would be awesome if you can  test them.

For testing please set debug mode to DEBUG_RC_INTERPOLATION

https://drive.google.com/drive/folders/1e0OIMWFTffFXDTUp0VvUKM1wDUM2kQ2t?usp=sharing

You can contact me on slack(nickname MD)

smasslich, 2018-04-25:
Thanks MD!  I'll give it a test.  Other than debug mode, should I also set set the following?
rc_interp = MANUAL
set rc_interp_ch = RPYT
set rc_interp_int = 14
set throttle_boost = 50
set throttle_boost_cutoff = 40

flint723, 2018-04-25:
I would keep the same settings you used previously. 

Also on my test build I cannot set debug_mode = debug_rc_interpolation so perhaps it's a typo?  I can set rc_interpolation.

flint723, 2018-04-25:
Whatever the test build is doing, seems like it made the spikes greater in amplitude and more often (more often 11ms apart instead of 22/33ms)

Log with 11ms and 19ms rc_interp_inteval:
[impulsercf3 thr interp on, test build, interval 11 and 19.zip](https://github.com/betaflight/betaflight/files/1949199/impulsercf3.thr.interp.on.test.build.interval.11.and.19.zip)
[diff impulsercf3 thr interp on, test build.txt](https://github.com/betaflight/betaflight/files/1949200/diff.impulsercf3.thr.interp.on.test.build.txt)

smasslich, 2018-04-26:
I'll test tomorrow AM and post the results.  Thanks!

kmitchel, 2018-05-13:
Any solution yet?  I'll try setting to auto, tomorrow.

etracer65, 2018-05-15:
I was able to replicate on the bench and have tracked the cause down to #5764 

Not sure what part of that PR is the problem, but I suspect some of the new alt-hold code in there might be erroneously affecting throttle - but just a guess at this point.

spatzengr, 2018-05-16:
Additionally, it appears #5764 was merged before Jenkins build no. 792.  A post by @theSkonk indicates he did not have the spikes in no. 792, but they re-appeared (different spikes) after build no. 830.

smasslich, 2018-05-16:
I could use a CLRacingF4 target hex.  Thanks!

etracer65, 2018-05-16:
Note that the problem isn't some "spike" out of nowhere.  It's actually the value for the next "step" in the throttle rcCommand that's erroneously being used for one PID loop cycle before being smoothed with the previous value subsequently.  In fact it's the same value that would occur if rc interpolation was turned off for throttle. It just looked like a spike because it was only used for one PID loop.  This is also why it varied in magnitude depending on the speed the throttle was changing.  The "step" would be taller.

PR #5905 (2018-05-18): Break out rc interpolation as a separate task and run before pid controller and motor update
Fixes #5759 

Fixes an issue with motor "spikes" when rc interpolation was enabled for throttle. The problem was that the smoothing was happening too late in the sequence and the earlier process `subTaskMotorUpdate()` would use the un-smoothed throttle value in the mixer if new rx data had come in between the last and current PID loop.  So the mixer would use the large step in the rc data for the first loop after new data came in causing the spike.

Also because the setPointRate was calculated at the e [... cut]
closed 2018-05-18 by mikeller