betaflight/betaflight #5763: FC setup screen issue when using OSD

DenJD, 2018-04-24:
My FC setup is BF 3.3.0 Omnibus F3 with OSD.

The right side of the FC setup screens (values) when using OSD is truncated on my EV100 and Vision+ goggles. The values are not visible at all in both NTSC and PAL modes. This may (probably) is something with the goggles but it would be good to have a solution or setting. These values would be visible If the they could be moved 5 (or so) characters to the left.

Is there a setting or a work around for this?

AndersHoglund, 2018-04-24:
You can try and adjust the CMS screen position on OSD by row or column using those parameters:

```
# get displayport_max7456
displayport_max7456_col_adjust = 0
Allowed range: -6 - 0

displayport_max7456_row_adjust = 0
Allowed range: -3 - 0
```

DenJD, 2018-04-24:
Anders,

That worked great!
Thanks much!

Dennis

From: Anders Höglund 
Sent: Tuesday, April 24, 2018 12:46 PM
To: betaflight/betaflight 
Cc: DenJD ; Author 
Subject: Re: [betaflight/betaflight] FC setup screen issue when using OSD (#5763)

You can try and adjust the CMS screen position on OSD by row or column using those parameters:

# get displayport_max7456
displayport_max7456_col_adjust = 0
Allowed range: -6 - 0

displayport_max7456_row_adjust = 0
Allowed range: -3 - 0

DenJD, 2018-04-24:
That worked great!
Thanks much!

closed 2018-04-24 by DenJD