betaflight/betaflight #5771: Have you ever considered using a recurrent neural network for filtering?

FabianIsensee, 2018-04-26:
Hey everyone,
in light of the current hype about filter methods: Have you ever considered using a recurrent neural network for filtering gyro traces? I think it could work really well and I would also be willing to experiment with training/designing such a network (I am more of a CNN kind of guy but should be able to do RNN as well). The big question is however: What is the computational budget that an F4/F7 has for filtering? Are a couple of matrix multiplications too much per update? And is there a database with tons of gyro data available (basically blackbox log with raw and filtered gyro data).
Regards,
Fabian

DieHertz, 2018-04-26:
Hi

There's no such database, actually getting logs when you really need them from users is hard :)
You could join our [Slack](https://slack.betaflight.tech) #test_feedback channel and start collecting logs there.

There's basically no budget on F4, and not much more on F7. If it could shed some of the existing filtering, some matrix algebra would be possible, but what is the dimensionality of matrices in question?

FabianIsensee, 2018-04-26:
HI,
I talked to a colleague of mine. The simplest network we came up with would need:
- 39x1 (vector) x 39x32 (matrix)
- 32x1 x 32x32
- 32x1 x 32x3
Additionally, we need to compute either the tanh or the sigmoid function for 64 scalar inputs
Is that beyond reasonable or could this work?
Regards,
Fabian

DieHertz, 2018-04-26:
Highly unlikely that we'll be able to do that on F4/F7 :)
Wait for H7 or Cortex-A

FabianIsensee, 2018-04-26:
Too bad :-/

DieHertz, 2018-04-26:
You'll have to train it offline anyway, so it's worth a shot, just no estimate on how soon we'll be able to implement it in FC firmware.
H7 will likely be available by the end of this year as I'm going to start working on it soon with other devs.

Keep in mind that we only have single-precision floating point support.

FabianIsensee, 2018-04-26:
Training is much more complicated than inference so I will have to do that offline anyways. I will probably do that in one of the big frameworks (pytorch). I would of course then have to implement the network in C from scratch and import the trained parameters.
single precision is no problem at all. All state of the art neural networks use single precision only. If it improves computation times, some people even go down to half or lower depending on the application

DieHertz, 2018-04-26:
If you'll need help with transferring it to C/C++, people at our Slack are really helpful and can lend you a hand :)
Have fun

ledvinap, 2018-04-26:
@FabianIsensee: My feeling is that it is not worth for realtime flight control. But It may be interesting as reference to compare common methods against ..

digitalentity, 2018-04-26:
@FabianIsensee interesting research project. If you can acheive good filtering for multiple sampling rates it may be even worth including in the code. I see several potential problems:

1. Training data set. You need noisy data + clean data. We only have noisy data, you need to create an off-line way of cleaning up the data without introducing phase delay.
2. Multiple sampling rates. We can sample gyro at any frequency from 500Hz to 32kHz. You'll have to create/train an NN for each sampling rate.
3. CPU resources. NN will be much more time consuming compared to current filtering implementations which only take a few multiplications/additions for each sample.

FabianIsensee, 2018-04-26:
If it is worth for realtime flight control is unclear, we will see once we have some results.

@digitalentity Your points are vary valid. I think I can train with traces from blackbox logs, but I will of course have to account for the phase shift. How I don't know yet (suggestions appreciated). Indeed, we would have to have separate networks for separate sampling rates. Alternatively we could also try to design something that can handle several sampling rates, but that may make it more computationally intensive. Like you said, CPU resources are scarce. Even if they are insufficient now, it might pay off in the long run to invest in research effort now ;-)

digitalentity, 2018-04-26:
@FabianIsensee you can probably use symmetric window FIR filter to cancel noise. The benefit of off-line filtering is that you can filter using future data and past data, while at run-time you only have past data to work with. Having future data will allow you to get a delay-free and noise-free traces to use as a training dataset.

If you manage to create a decent NN for filtering gyro data I'll be happy to test it and maybe merge in INAV which doesn't require high-rate gyro updates (1-2 kHz is most commonly used rate)

FabianIsensee, 2018-04-27:
@apoc Thanks a lot for sharing. I have somme comments about your approach that should allow you to improve it substantially:
1) using linear activation functions within the network is a mistake. If you stack dense layers without activations (which is what linear activation basically means) you are nesting linear functions. Nested linear functions can in fact be expressed by a single linear expression. The network architecture you designed is not any more powerful than linear regression. Use ReLU, tanh or sigmoid between layers (final output must be linear though)
2) if you decide against using an RNN (which is totally fine) then you should consider using stacked 1d convolutional layers. They use less parameters and overfit less. It has been shown in the literature that 1d convolutions can perform just as well as an RNN, so this is a very interesting thing to try. Also you can train 1d convolutional neural network in a fully convolutional way allowing it to train more efficiently.  On the other hand, 1d convolutional neural networks are have a symmetrical field of view in time. If you train it naively on the data then it will also look at future data which is not what we want here. We want a filtered gyro value output conditioned on the last X time steps
3) like you said in the video, a lpf with a cutoff at 30 not ideal. I think propwash is around 100hz!? you would not want to filter that.

How did you manage to load the BBL in python? I would like to also do some experiments this weekend ;-)

DieHertz, 2018-04-27:
There are `blackbox-tools`, allowing you to output CSV

apoc, 2018-04-27:
Yes, this is what I have used ;)

apoc, 2018-04-27:
@FabianIsensee 

Thanks for input, I can try the suggestions, but I have feeling that in first tries I had ReLU/tanh there and linear at output and the result was very similar to all linear anyway I can retest it today :)

FabianIsensee, 2018-04-30:
[image]
Here is a little sneak peek. Green is raw gyro, blue is my 'ground truth' (raw gyro filtered with symmetrical FIR and cutoff 100hz), red is my RNN output

digitalentity, 2018-04-30:
Looks promising, especially if your RNN does not look into future data :)

digitalentity, 2018-04-30:
How intensive is it on CPU?

FabianIsensee, 2018-04-30:
Of course it does not look into future data. That is the whole point ;-) What I did to make it respond fast is to give it the rc inputs as feature as well. That way it can anticipate gyro movements and respond rapidly. You can actually see that is has a phase delay for unanticipated movement (like propwash) while being super responsive to anything that is caused by the quad moving in response to stick inputs.
The output of the RNN is smoother than the gt. I achieved that by using an additional loss term that is similar to the total variation loss.
As of now it is too intensive on the CPU to run on current hardware. This is just a proof of concept that should demonstrate what is possible. I will run a series of experiments where I make the network gradually smaller and see how that affects the quality of the filtering. What you see in the graph was generated from a RNN that uses one GRU layer with 32 hidden units. Prior to the GRU there is a linear layer with ReLU activation and the output is generated by another linear layer. The linear layers have weight matrices of shape 7x32 and 32x3, respectively. The GRU has 2 weight matrices, each of shape 96x32.
The traces in the plot originate from a test case by the way. The network has not seen this trace in training.

digitalentity, 2018-04-30:
Feeding stick input into RNN is a bad idea. This will make phase delay non-stop constant and will actually make tuning less intuitive. Gyro as a function of stick input is a function of tuning. You change tuning - you also invalidate the training of your RNN, because relation between sick and gyro will change

FabianIsensee, 2018-04-30:
[image]

I would disagree that it is a bad idea per se. Of course the reaction time of the quad depends on tuning, but including them could help provided there is enough training data. For simplicity I now removed the stick input to the RNN to rule out tuning related issues (especially because my 'datset' is very small and only contains data from 4 quads).
One other problem I encountered is my 'ground truth' which I train on. When plotting the filtered gyro traces form the log and comparing them to what I filtered in python, the betaflight filtered values look much better! For this reason I now retrained with the filtered gyro data as reference. When comparing the result on the test set, my filtering does worse than betaflight filtering (now green = raw, blue = filtered gyro form the log, red = my prediction). This particular quad should have only a PT1 filter (which from my understanding is just a moving average filter) and dynamic notch enabled. It may have an additional notch filter at ~400 hz.
I would very much like to explore some more options/architectures. For that I will need really high quality ground truth data. Since I really don't have any background in filtering, could someone please provide a short python snippet with which I can get a really accurate ground truth trace for training?

DieHertz, 2018-05-02:
PT1 isn't just a moving average, it's a IIR filter, while moving average is FIR.
PT1 is a filter with transfer function in form of `w / (s+w)`

rav-rav, 2018-05-02:
@FabianIsensee Here are some of the betaflight filters in python: https://github.com/rav-rav/betaflightCalc

FabianIsensee, 2018-05-02:
All I found about PT1 filter (without loooking into betaflight code) was this link here:
https://infosys.beckhoff.com/english.php?content=../content/1033/tf5000_tc3_nc_ptp_10_axes_pfm/3625967627.html&id=

Where this: `x = I * x_n + (1-I) * x_a` looks like a moving average with some weighting while I is parameterized by something that looks like your transfer function. 

Regardless of filter definitions, how would I generate the best possible filtered traces to train my network on? Low pas filter in fourier domain? Is there something more elaborate I could use? If so, is there a reference implementation?

Honestly I am pretty blown away by the smoothness of the blue curve (betaflight filtering) in my plots. How can it respond that fast to an actual change in inputs? It looks like it is sometimes rising before the actual raw gyro rises? I would have expected to see a 3-5ms delay. Did I do something wrong in my data preprocessing?

FabianIsensee, 2018-05-02:
@rav-rav Thanks a lot! Which one should I use to generate the best possible trace? The good thing about the logs is that I can look into the future whcih should enable even better filtering. can the filters in that repo also use this?

DieHertz, 2018-05-02:
>Where this: x = I * x_n + (1-I) * x_a looks like a moving average with some weighting while I is parameterized by something that looks like your transfer function.

The catch is `x_a` is not previous input value, but rather previous output value.

rav-rav, 2018-05-02:
The code is pretty much the same as in betaflight, so it won't use any future data. There are some examples on how to cascade different filters in filterAnalyzer.py.
The dynamic notch filter does remove quite a lot of noise though, which is not available in python.

You could also look into blackboxLoader.py and compare it to your approach of loading logs, if you are not sure it's correct.

It's hard to tell what a good filtering setup would be, but maybe a definition of a good gyro trace could be helpful? Personally I'd expect the result to have no frequencies above 150-200Hz.

FabianIsensee, 2018-05-02:
@DieHertz I was aware of that and thought that was what a moving average filter does. But like I said I have no background in filtering so I'll go by your word ;-)

@rav-rav I will check it out and also share my code soon if you are interested. 

>  I'd expect the result to have no frequencies above 150-200Hz.

So I guess it's going to be an FFT -> low pass filter. I will report back once I have more results

bkeinert, 2018-05-03:
@DieHertz what @FabianIsensee  is referring to is also correct, more explicitly he's referring to an exponential moving average, which is simply an IIR filter (See https://en.wikipedia.org/wiki/Moving_average#Exponential_moving_average).

While we are at it, what about training filter parameters?

closed 2018-06-10 by stale[bot]