betaflight/betaflight #5772: MOTOLAB target has SPI Flash M25P16 defines in target, breaking SPI MPU-6000

fujin, 2018-04-26:
The board does not have SPI flash (see [pictures](https://www.rcgroups.com/forums/showpost.php?p=32330479&postcount=2)) and the recent addition of M25P16 defines in master breaks MPU6000 SPI gyro on the target. 

I noticed that the SPI/CS are the same:
https://github.com/betaflight/betaflight/blob/master/src/main/target/MOTOLAB/target.h#L56-L57
https://github.com/betaflight/betaflight/blob/master/src/main/target/MOTOLAB/target.h#L89-L90

Strangely enough, the craft allows arming while in this state. 

It would be good to:
- Understand why the gyro on target breaks with these defines present, specifically allowing arming
- Remove the incorrect defines from the target

Refs #5770, #5683 - introduced to MOTOLAB at same time.

Thanks to @ctzsnooze for reporting this. FYI @MotoLab.

MotoLab, 2018-04-26:
One of the boards using the MOTOLAB target (MOTOF3) does have an M25P16 for data logging. All the previous versions have worked with or without the flash on board. I don't want that removed from the target.

DieHertz, 2018-04-26:
How would you explain CE being PB12 for both gyro and dataflash?
It's not about what you want, it's about fixing the issue

ctzsnooze, 2018-04-27:
Thanks @fujin - I can confirm that:

`resource FLASH_CS NONE`

allows gyro to work on motolab targets... as does a custom compile as suggested by @DieHertz where in target/motolab.h comment out these lines as follows

`//#define USE_FLASHFS
//#define USE_FLASH_M25P16`

jflyper, 2018-04-28:
@ctzsnooze This firmware from #5776 should work.

[betaflight_3.4.0_MOTOLAB.hex.zip](https://github.com/betaflight/betaflight/files/1957182/betaflight_3.4.0_MOTOLAB.hex.zip)

DieHertz, 2018-04-28:
@ctzsnooze I suggested you to try `resource flash_cs none` and abandon custom flashing in our Slack :)

ctzsnooze, 2018-04-29:
I flew standard code with ```resource flash_cs none``` yesterday, and it worked fine.

ctzsnooze, 2018-04-30:
Can't bring good news with that hex, unfortunately.  Wouldn't boot.  Needed to boot loader it, tried a couple of times, then flashed 3.4 from yesterday and that booted fine.  Sorry.

jflyper, 2018-04-30:
Can't happen...

How about this?
[betaflight_3.4.0_MOTOLAB.hex.zip](https://github.com/betaflight/betaflight/files/1962503/betaflight_3.4.0_MOTOLAB.hex.zip)

If this doesn't work, I need to do some deep debugging.

ctzsnooze, 2018-05-01:
*that's better*  :-)
works 'out of the box', perfectly, both gyro and acc fine.
thank you!

jflyper, 2018-05-01:
@ctzsnooze Thanks for testing! 😄

PR #5776 (2018-05-01): FLASH: Handle overloaded SPI CS pin
A new `flashInit`, introduced by #5683 which de-init designated CS pin when it fails to detect a valid flash chip, broke some target that has flash CS pin overloaded with other SPI devices. MOTOLAB (overloaded with MPU gyro) and FF_FORTINI (overloaded with MAX OSD chip) are known to be affected by this behavior.

Older flash device detection did not de-init, which "accidentally" allowed actual resident device on the CS pin to continue to work.

This PR proposes a limited but simple solution to prevent `flashInit` from touching a CS pin that is already initia [... cut]
closed 2018-05-01 by jflyper