betaflight/betaflight #5773: Low battery shakes quad

jorksx, 2018-04-26:
I have flewn whole winter without any problems, even when OSD is flashing the LOW BATTERY warning. even then it flew fine.
Now when I see Low battery, at the same time quad shakes in the air.
If I full throttle and the same time low battery comes up. quad starts going nuts. If I fly normally no problems. the quad always rolls left and right very very fast.
its like there is desync or something.

I have replaced every ESC ( dys 20A )
no effect
FC is DYS 4-1

etracer65, 2018-04-26:
Try turning off "Vbat PID Compensation" on the PID Tuning tab.

jorksx, 2018-04-26:
it is already off

etracer65, 2018-04-26:
Then you should provide a `diff` of your settings (as requested in the issue creation instructions).  Also provide a blackbox log showing the problem.  Indicate the spot in the log where the problem is observed so we don't have to guess.

jorksx, 2018-04-26:
okey I'll go fly soon and I try to remember to enable blackbox recording.

McGiverGim, 2018-04-26:
Maybe is a stupid answer, but, maybe the beeper is affecting the motors? Some timer stuff or similar. Try disabling the beeper feature.

DieHertz, 2018-04-26:
Active buzzers don't use timers

jorksx, 2018-04-26:
What timers where?
I'll try disable buzzer. 

But funny thing is I did not change anything and the problem just appeared

jorksx, 2018-04-26:
is my fc dead almost?
because I did blackbox reading, now I downloaded the file and tried to open it with blackbox explorer and it says, logging was paused, no data
I deleted everything before I started blackbox recording so there is only one flight recorded.

jorksx, 2018-04-26:
from dropdown there is like 20 of them and all of them are logging paused, no data recorded.
And from betaflight modes, aux 4 is beeber and blackbox.

when I activate aux 4, beeber starts beebing so I assume that blackbox to.

jorksx, 2018-04-26:
https://ufile.io/jdt4k

there is file for my blackbox file, timeline 00:10:33
you can see when I full throttle, motors are doing something wierd.
I dont know how to read blackbox or gyro info so can anyone tell me whats wrong with it?

and this is only when OSD starts telling me "LOW battery"

etracer65, 2018-04-26:
Can you please go to the CLI and do post the results of a `diff` as requested?

jorksx, 2018-04-26:
sorry, here https://ufile.io/thk3r

DieHertz, 2018-04-26:
Check the value of `sdcard_dma`, should be off

jorksx, 2018-04-26:
# sdcard_dma
Unknown command, try 'help'
# sdcard_dma=off
Unknown command, try 'help'
#

etracer65, 2018-04-26:
The logging rate in your blackbox log is too slow to tell anything meaningful about the filtering, noise or even really the PIDs.  You should lot with a rate of at least 1KHz.

It looks to me though that your battery is just worn out.  On your punch outs the battery is dropping down to 3.0v/cell which is really limiting the available response from your motors likely leading to the oscillation.  Even as soon as you took off and were basically hovering it was only showing about 3.75v/cell which is a battery that's about 75% depleted.

jorksx, 2018-04-26:
blackbox is done in 1kHz ? do I need to change it to 2?

Yes the battery is dropping, but its doing this with ALL of my batteries. and when I flew in the winter, there was no problem like this.

etracer65, 2018-04-26:
You need new batteries.  Each subsequent punch out in the log shows more oscillation as the battery is getting weaker and this is reflected in the vbat trace.

jorksx, 2018-04-26:
5 of the batteries are 2 weeks old and I fly maybe 4 times them. no way they are bad already.

etracer65, 2018-04-26:
Well then you have some other problem in your power system.  Maybe a bad PDB, wiring etc.  Possibly a bad motor that's drawing too much power, don't know.  The vbat trace doesn't lie.  Your voltage is dropping too low on punch outs.  Maybe you're not charging your batteries properly as it was only showing 3.83v/cell when you armed.

Anyway, I recommend you take this disussion about your hardware setup and/or batteries to the rcgroups thread as that's the more appropriate place.  I'm going to close this because it's not related to a problem with the firmware.

rcgroups: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/page3658

closed 2018-04-26 by etracer65

DieHertz, 2018-04-26:
`set sdcard_dma`, not just `sdcard_dma`

jorksx, 2018-04-26:
# set sdcard_dma
Invalid name

# set sdcard_dma=off
Invalid name

# set sdcard_dma = OFF
Invalid name

richard-scott, 2018-04-26:
I get the same on an AnyFC F7 board when I get to the bottom of my battery (like < 10% left)... I have no VBat compensation on either. I always thought it was just that there was not enough juice for it to fly and not a bug. I can get a BB log if required.

jflyper, 2018-04-26:
There's nothing we can do with dud batteries and/or poor power systems.

jorksx, 2018-04-26:
I still think that batteries are not the problem. It first says low battery, after it gets really low then it says "Land NOW"

2 months ago even when it said land now, there was no shaky shaky problem.

jflyper, 2018-04-27:
I suspect your charger is broken, destroying every battery you try to charge, regardless of old or new.

Anyway, it’s not a software problem for sure.