betaflight/betaflight #5774: Unify the Blackbox log file name

McGiverGim, 2018-04-26:
Taking a look to the configurator and to the firmware, I see that the name of the Blackbox files are different when downloading a flash or when writing to the SD.

When writing to the SD, the names are in the form `LOGXXXXX.BFL` but the code of the configurator, if I'm not wrong, when downloading a flash the name by default is in the form: `[BFCF_]BLACKBOX_LOG[_NAME]_YYYYMMDD_HHMMSS.BBL`

I don't have a FC with integrated flash to test it, I don't know if someone can verify the name.

But I think it will be a good idea to unify both names/extensions.

flint723, 2018-04-27:
Default file from flash, recent 3.4 CI build
[image]

jflyper, 2018-04-28:
There's one more thing to remember; FAT emulation is not capable of handling long file names...

McGiverGim, 2018-04-28:
Are you saying we are limited to 8 characters long under fat emulation? This reduces a lot the possibilities...

mikeller, 2018-05-04:
@McGiverGim: Yes. I looked into this before, but the fat emulation can only do 8.3 file names. Also, in a lot of cases there is no real time clock on the flight controller, so adding a (helpful) timestamp would not work even if we had long file names. For now I think what we've got is the best we can do without throwing more flash space at it, and making it unattractive for F3.

McGiverGim, 2018-05-04:
Yes, you are right. I was looking at modify that but I think it does not worth it with the actual limitations.

closed 2018-06-10 by stale[bot]