betaflight/betaflight #5777: I2C compass and barometer not recognized with Super_S F7 20x20mm

andrejpodzimek, 2018-04-27:
I've built a drone with [this 20x20mm F7](http://www.buzzhobbies.com.au/mini-f7-flight-controller-20mm) board and soldered my favorite [compass](https://www.banggood.com/3Pcs-GY-282-HMC5983-High-precision-High-sensitivity-Temperature-Compensation-p-1144414.html) and [baro](https://www.banggood.com/3Pcs-MS5611-GY-63-Atmospheric-Pressure-Sensor-Module-IICSPI-Communication-p-1122870.html) together onto the SDA and SCL pads. (These two devices work fine together on a single I2C bus on one of my F4 builds.) Both are powered from 5 V.

Neither the compass nor the baro are recognized and I always get 6 I2C errors. SDA under oscilloscope is very concerning:

[image]
[image]

This^^^ looks like there's not enough pull-up resistance, so some of the devices can't get the bus down to 0. The resistance between 5V and SDA (and 5 V and SCL) is only about 1.5kΩ. That's not enough, right? On my working F4 setup, the resistance is 1.95kΩ.

There used to be a CLI option in Betaflight to enable / disable internal pullup resistors, but I can't find it any more. Is this still configurable?

What else could be causing such a poor signal? Any configuration options I could try? If I need to physically remove a pullup resistor from the compass, the baro or (oh horror) the FC, how do I find it on the board?

McGiverGim, 2018-04-27:
As far as I know, there is no CLI command, but some defines:
https://github.com/betaflight/betaflight/blob/02efd46833155f43b77a15f8ebe795484898d488/src/main/pg/bus_i2c.c#L165-L175

Try to define `USE_I2C_PULLUP` for your target and build the firmware.

jflyper, 2018-04-27:
Pull-up is too strong, and it is pulled up to 4.4V, which is bit high (and weird). There seems to be substantial capacitance on the bus also.

In summary, the bus is broken. I suspect there are pull-ups already on the FC, and I2C modules has their own pull-ups. There might even be a capacitor installed across signal and ground.

closed 2018-04-27 by jflyper

andrejpodzimek, 2018-04-27:
I forgot to mention one thing: I compiled the firmware with an additional timer definition, because I wanted to have LEDs on UART1. The thing doesn't have any dedicated LED pad. My LEDs work fine with this:

```c
DEF_TIM(TIM1, CH2, PA9, TIM_USE_LED, 0, 0), // LED on UART1 TX
```

But I2C malfunctions the same way without that extra timer, so it's probably not causing the problem.

@McGiverGim `USE_I2C_PULLUP` is not defined for `ANYFCF7`, so it defaults to `false`. What are the semantics of that flag? (I'm lazy to parse the bitmaps in `obj/main/ANYFCF7/drivers/bus_i2c_hal.i`.) Does `true` mean _activate some internal pullups_ or does it stand for _don't activate the internal pullups, because there are external ones already_? I assumed the former, so I thought I'd have to _unset_ it, but there's nothing to unset for `ANYFCF7`.

@jflyper So should I just try to identify resistors or capacitors on the PCBs that have continuity towards SDA/SCL on one end? (Not sure about the other end; they shouldn't go directly to 5V (well, the >4V on the bus is weird, indeed). The 3.3V power pad for Spektrum could be a continuity reference, maybe...?) But I guess I'd need a hot air system to desolder such tiny resistors. :scream: I'm also surprised about the capacitance on the bus; the only capacitor I installed is a big one on the battery input, unrelated to low-voltage things.

jflyper, 2018-04-28:
Another comment.  No offense, just facts.

The baro unit is probably made to pull the bus to supply voltage, and has level shifters for this purpose. The unit was designed to interface with 5V systems, not for 3.3V system like our FCs.

jflyper, 2018-04-28:
You should disconnect i2c devices before poking around.

andrejpodzimek, 2018-04-28:
@jflyper I see. I'll try to get more suitable devices then. Both of these (i.e., same devices from the same shipment) work fine on my F4 build, but I guess that's just by chance.

jflyper, 2018-04-29:
I2C becomes much easier and reliable if things are planned and wired correctly, which is never accomplished when modules are combined and connected in some random way.

andrejpodzimek, 2018-04-29:
Sure, it would be great to have a pre-built GPS+baro+compass that fits on a 3" drone, but I just can’t find it anywhere, which is why it always ends up with trial-and-error soldering.

jflyper, 2018-04-29:
Don't start soldering unless you are confident about the consequences. If not, spend time creating a schematics by reverse engineering; those bobs are relatively simple. Evaluate the schematics and various components. Do the same around the pins and pads you are going to use on FCs.

We don't need any luck as we are not gambling.

andrejpodzimek, 2018-05-01:
[image]

So we have (or should have) schematics and we definitely don't gamble, right? :stuck_out_tongue:  Honestly, when I get a new board, I don't go over every single pad to check whether there's continuity towards the right MCU pin. But that's exactly what I should have done in this case. I just blindly assumed that the schematic was correct. [Here's the schematic from the seller's E-shop:](http://www.buzzhobbies.com.au/mini-f7-flight-controller-20mm)

[image]

**SDA and SCL are reversed in the schematic.** That's the first problem. Having tried all 4 possible I2C devices without success, I thought that the I2C pads might lead to wrong MCU pins. So I looked at the pinout [on page 45 here](http://www.st.com/resource/en/datasheet/stm32f745ie.pdf) and found out that `PB10` (SCL) and `PB11` (SDA) were indeed swapped in the schematic.

The second (and more obvious) problem was the voltage. The devices say 3V&ndash;5V in the specs, but there's a problem on the board: For some reason the 5V BEC and I2C don't get on well. (Which I only discovered with the I2C desoldered.) The resistance between 5V and SDA/SCL on the board was 32kΩ, but there was a very substantial capacitance between the two. Not sure why. That must have caused (or contributed to) the poor signal quality problem. So I tried the 3.3V pad (for Spektrum; normally unused in my case) and it had a steady 1.8kΩ of resistance towards SDA/SCL and no obvious capacitance issues.

Now the overall pullup resistance is only around 1kΩ. (That's because the board now contributes 1.8kΩ instead of 32kΩ to the "sum".) Well, as long as it works, I don't fancy poking into it again.

jflyper, 2018-05-01:
Congrats for findings and meta-findings! 😀