betaflight/betaflight #5779: Can I interact with Betaflight CLI through a linux terminal or windows command prompt?

klleww, 2018-04-27:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

AndersHoglund, 2018-04-27:
What terminal program are you using? I have no problem connecting with Putty. Please not that you have to initiate cli with a "#" char input. Then also, why selecting cli as the interface for scripting? Checkout how BFC connects via MSP instead.
[image]

neilbalch, 2018-05-03:
@AndersHoglund  Out of curiosity, what protocol and config did you set in PuTTY to connect to the FC?

AndersHoglund, 2018-05-03:
Serial COMx port, 115200.
[image]

avr39-ripe, 2018-05-03:
@AndersHoglund Man! Thank a lot for "you need to initiate CLI with #" ! Can't figure out it for a long time..