betaflight/betaflight #5780: osd Display Port FC_VERSION_STRING / CMS_STARTUP_HELP_TEXTN does not display

fujin, 2018-04-28:
Observed while testing #5776.

When powering up craft, LOGO is seen as per normal, rest of screen is blank, e.g. no FC version, no CMS help text on how to enter CMS menu. 

Looks like this code is not working as intended: https://github.com/betaflight/betaflight/blob/master/src/main/io/osd.c#L1003-L1017

etracer65, 2018-04-28:
Related to #5405 

Comments from @hydra indicated that this wasn't directly the cause, but rather exposed weaknesses in the max7456 initialization that were the underlying cause.

etracer65, 2018-04-30:
So not a solution (I don't think), but commenting out https://github.com/betaflight/betaflight/blob/fba59c32b6eec484e2106172ffa28673327f72c7/src/main/io/osd.c#L1019 fixes the problem.  But I have no idea what this would do to other displayPort types.

hydra, 2018-05-01:
yeah don't do that, you'll break osd over MSP if you do.

etracer65, 2018-05-01:
Okay, I've identified the cause but need to discuss the best solution.

Turns out this has nothing to do with DMA or even how the MAX7456 is initialized.  The cause is how `max7456DrawScreenSlow()` is refreshing the display.  It's putting the chip into "auto-increment mode" to more efficiently refresh all the character data.  Auto-increment mode just requires a starting position and then each byte sent moves the character position automatically.  This saves having to re-position the address each time (saves 2 spi writes per character).

From the datasheet:

> This mode is disabled by writing the escape character 1111 1111.

Turns out that the last byte written for the logo is 0xFF.  So when `max7456DrawScreenSlow()` reaches that byte in the buffer the chip turns off auto-increment mode and the remainder of the screen buffer just continues to update the same position on the screen (the last character position of the logo).

So, a couple of ways to fix.  First I'm not sure if character 0xFF should be valid at all.  It seems to be treated specially as the "escape" character by the MAX7456.  So maybe the flaw is in how the logo is drawn as it draws every character from 160 to 255 which are expected to contain the graphic chunks for the logo.  In /io/osd.c the following change fixes the problem:

```
static void osdDrawLogo(int x, int y)
{
    // display logo and help
    int fontOffset = 160;
    for (int row = 0; row < 4; row++) {
        for (int column = 0; column < 24; column++) {
-            if (fontOffset <= SYM_END_OF_FONT)
+            if (fontOffset < SYM_END_OF_FONT)
                displayWriteChar(osdDisplayPort, x + column, y + row, fontOffset++);
        }
    }
}
```
But I can't tell if we're losing part of the logo (or if character 0xFF is valid).

Alternately if character 0xFF is really valid and needs to be displayed, then in /drivers/max7456.c `max7456DrawScreenSlow()` will have to be changed to either not use auto-increment mode or some more intelligence will have to be built in to use auto-increment and skip any 0xFF characters and then make a second pass to update just those with direct addressing.

So it all comes down to whether 0xFF is a valid character we need to display.

Note that all this only applies when a full refresh is taking place like after initialization or during a video mode change - it's not part of the normal OSD updates as they're handled separately.

Related but not a problem: How does the font in the MAX7456 get initially loaded the first time a flight controller is flashed?  Somehow we must be loading a font into NVM on the MAX7456 with the logo characters but I can't seem to find where that's happening.  The NVM update in msp.c only seems to update the display font range from the font manager in the configurator.

etracer65, 2018-05-01:
And looking at osd_slave.c it appears this problem was addressed:

```
static void osdDrawLogo(int x, int y)
{
    char fontOffset = 160;
    for (int row = 0; row < 4; row++) {
        for (int column = 0; column < 24; column++) {
            if (fontOffset != 255) // FIXME magic number
                displayWriteChar(osdDisplayPort, x + column, y + row, fontOffset++);
        }
    }
}
```
So I'm assuming we want to handle the same in normal osd.c.

jflyper, 2018-05-01:
Ah, yes, your observation is correct.

For the solution, I would do it in `drivers/max7456.c` to dodge 0xff, as this is a special char for MAX.
```
527 void max7456WriteChar(uint8_t x, uint8_t y, uint8_t c)
528 {
529     screenBuffer[y*CHARS_PER_LINE+x] = (c == 0xff) ? ' ' : c;
530 }
```

Ultimate solution is to detect 0xff during display memory writing, but it will be too expensive to detect the termination of the incremental mode (by 0xff), write 0xff with independent write mode and then restarting the incremental mode.

Because of this, we might decide 0xff as an invalid char for displayPort devices. We can then dodge 0xff inside `displayPortWriteChar`.

etracer65, 2018-05-01:
I don't see any problem with avoiding the 0xFF characters during the initial auto-increment phase and then coming back to draw them with direct addressing.  It seems expensive but it's not that bad.  We don't expect there to be many 0xFF characters.  Also we need to realize that this code only runs during initialization and events like video mode changes (PAL to NTSC, etc.) so the impact during normal OSD operation is none.  I have a working version so I'll put together a PR.  That way we don't have to worry about avoiding the 0xFF character.

PR #5792 (2018-05-03): Avoid 0xFF character when writing to MAX7456 in auto-increment mode
Fixes #5780 
Fixes #5791 

The 0xFF character is used as an "escape" character by the MAX7456 when using auto-increment mode. The last byte of the characters used for the logo is the 0xFF character and this was causing the version and CMS instructions to not display.

Changed the logic to avoid the 0xFF character during the auto-increment phase and added a second pass if needed to update those characters in direct addressing mode.

Exposed (but not caused) by #5405.  Does not need to be applied to 3.3 as the problem began in 3 [... cut]
closed 2018-05-03 by mikeller