betaflight/betaflight #5784: Beeper pattern "----- — -" and disarming on quick yaw moves

andrejpodzimek, 2018-04-28:
On my `OMNIBUSF7V2` build (8", [T-Motor Navigator 3110 780KV](http://store-en.tmotor.com/goods.php?id=336), 6S, [HGLRC 45A/60A ESCs](https://www.hglrc.com/collections/escs/products/hglrc-t-rex-60amp-60a-blheli_32-3-6s-esc-dshot1200-for-rc-drone-fpv-racing-multi-rotor)) I wanted to double-check that [this scary event](https://www.rcgroups.com/forums/showthread.php?2957356-F80-Wraith32-Plus-went-up-in-smoke/page7#post39531360) can no longer happen after a rebuild (having replaced the dangerous Airbot Wraith32 ESCs and under-powered [F80](http://store-en.tmotor.com/goods.php?id=314) motors (also unfit for the props)). (Better to check that on the floor than in the air.)

I made a few sharp yaw wobbles with throttle at 0. Nothing bad happened to the motors and/or ESCs, no overheats, no ESC restarts. So things looked fine initially.

However, moving the yaw very quickly to the edge (at zero throttle) yields an immediate **disarming** and a beeper pattern of "----- — -" (five short, one long, one short), which I could not identify [here in the source code](https://github.com/betaflight/betaflight/blob/master/src/main/io/beeper.c) (or I just overlooked it).

What does that beeper pattern mean? Where is it defined? Why does my drone disarm due to sharp yaw wobbles on the floor? Will this happen also in the air?

I'm using an arming channel, so it can't be the case that moving the (mode 2) left stick into the corner could arm/disarm. Also, both left and right wobbles do cause the same thing. And it only happens when the movements are quick, not when the stick is moved to the edge slowly. The drone is not tethered and can therefore spin freely on my slippery floor.

wind0r, 2018-04-28:
bf 3.3 detects that something is wrong and disarmes since you dont have props on(atleast i hope you dont have them on) the quad cant stablize(do the yaw movment as expected) and disarms

andrejpodzimek, 2018-04-28:
@wind0r Of course I do have props on. How would one “load-test” the motors without props?

Maybe it’s the friction against the floor that confused the FC. With yaw wobbles at zero throttle (which is part of the experiment I wanted to carry out to check whether I get a fire again), the quad doesn’t get much lift and has to overcome lots of friction to actually turn on the floor. Maybe that confuses the FC into thinking it’s in a runaway situation. The thing has 1.54 kg, so there is substantial friction against the floor.

I was somewhat worried whether this could also happen in the air. @etracer65, based on what you say, I assume the answer is no. I’ve just tried a few sharp wobbles in flight / ground effect and no disarming occurred.

Still one question remains: Where in the code is this particular beep signal defined?

etracer65, 2018-04-28:
The documentation of the arming disabled beep patterns is here:
https://github.com/betaflight/betaflight/wiki/Arming-Sequence-&-Safety

If you really want to see the code, look here: 
https://github.com/betaflight/betaflight/blob/master/src/main/io/beeper.c#L308-L344

andrejpodzimek, 2018-04-28:
Thanks for the reference. I had been looking only at the constat arrays above that and mistakenly assumed that those were all available beeper signals. Now I've found [this call site](https://github.com/betaflight/betaflight/blob/02efd46833155f43b77a15f8ebe795484898d488/src/main/fc/fc_core.c#L385) that most likely generates the signal.

Anyway, this doesn't look like a bug, so I'm closing it. Thanks for the clarification.

closed 2018-04-28 by andrejpodzimek