betaflight/betaflight #5785: Smartport VFAS timing out and causing continuous sensor losts in opentx

jirif, 2018-04-28:
With current 3.4 master is not posible to use VFAS telemetry data because its not refreshed within opentx sensor timeout, so It causing continuous  sensor lost messages and unacceptable delay.

Also some invalid sensor data are sending. I tracked the source of the problem to #5581. 

Reverting #5581 and associated #5651 fixed both problems.

After short review of #5581 i have found than invalid sensor data are sent because of unchecked NULL pointer refernces in FSSP_DATAID_RPM  and FSSP_DATAID_TEMP ...

```
# version
# Betaflight / MATEKF722 (MKF7) 3.4.0 Apr 28 2018 / 20:03:42 (d4adc0561) MSP API: 1.39

# name
name JIRI

# resources
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 A08

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1000 1100 0
aux 1 0 1 1400 1500 0
aux 2 13 3 1100 1225 0
aux 3 19 3 1025 1100 0
aux 4 19 3 1150 1225 0
aux 5 26 1 1000 1100 0
aux 6 28 1 1000 1100 0
aux 7 35 1 1400 1500 0

# adjrange
adjrange 0 0 2 975 1025 18 0 0 0
adjrange 1 0 2 1125 1175 15 0 0 0
adjrange 2 0 2 1300 1350 9 0 0 0
adjrange 3 0 2 1025 1075 19 0 0 0
adjrange 4 0 2 1200 1250 16 0 0 0
adjrange 5 0 2 1350 1400 10 0 0 0
adjrange 6 0 2 1075 1125 20 0 0 0
adjrange 7 0 2 1250 1300 17 0 0 0
adjrange 8 0 2 1425 1475 11 0 0 0
adjrange 9 0 2 1475 1525 0 4 0 0

# rxrange

# vtx

# rxfail

# master
set gyro_use_32khz = ON
set acc_hardware = NONE
set mag_hardware = NONE
set min_check = 1030
set max_check = 1970
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT600
set beeper_dshot_beacon_tone = 1
set small_angle = 180
set runaway_takeoff_prevention = OFF
set osd_warnings = 127
set osd_rssi_alarm = 50
set osd_vbat_pos = 2092
set osd_rssi_pos = 2088
set osd_tim_1_pos = 2465
set osd_tim_2_pos = 2098
set osd_flymode_pos = 2520
set osd_throttle_pos = 2273
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 2497
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 2293
set osd_pid_roll_pos = 321
set osd_pid_pitch_pos = 353
set osd_pid_yaw_pos = 385
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2281
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_alt = ON
set vtx_band = 5
set vtx_channel = 4
set vtx_freq = 5769
set vcd_video_system = 1
set vcd_v_offset = -10

# profile
profile 0

set vbat_pid_gain = ON
set anti_gravity_gain = 4000
set dterm_setpoint_weight = 130

# rateprofile
rateprofile 3

set roll_rc_rate = 201
set pitch_rc_rate = 201
set yaw_rc_rate = 201
set roll_expo = 78
set pitch_expo = 78
set yaw_expo = 78
set roll_srate = 57
set pitch_srate = 57
set yaw_srate = 57

# 
```

pulquero, 2018-04-29:
Thanks, I'll do a PR and drop a link here.

pulquero, 2018-04-29:
Please try pulquero/betaflight:esc-telemetry branch. I don't believe there is any missing NULL checks since ESC_SENSOR_COMBINED returns a pre-allocated struct.

mikeller, 2018-05-03:
Fixed by #5785.

commit 48093e44 (2018-05-03): Merge pull request #5786 from pulquero/esc-telemetry

Improved distribution of telemetry bandwidth over sensors (fixes #5785).
commit f2f51061 (2018-05-03): Improved distribution of telemetry bandwidth over sensors (fixes #5785).
closed 2018-05-03 by mikeller