betaflight/betaflight #5788: Strange flip behaviour

stevo1957, 2018-04-30:
My quad is running a Naze (v5?) board with Betaflight V 2.9.1 under configurator 2.9.0.

Everything works well except when I do a fast (as in stick movement) forward or reverse flip, the quad changes direction as it comes out of the flip. If I do slow flips, no problems, quad will retain the direction it is facing when the flip is initiated.

I can do fast or slow sideways flips with no issue.

I am just wondering where I start looking in terms of changes to correct this.

I have air mode permanently enabled and have the problem in horizon mode. I have not tried in rate mode as aren't a good enough pilot yet to fly without stabilisation assistance.

Any suggestions?

AirBreak69, 2018-04-30:
Sorry, no suggestions. But I can confirm, that this behavior can occur with F4 boards and latest firmware, too.
I mostly fly in horizon mode too, due the my own limitations.
I guess, the DEVs fly in acro/rate mode - so the horizon mode effects get "out of sight".

I would like to state here another possible issue in horizon mode, that may be based on the same effects.
Lets call it "tilted horizon after tilted stable flight with acceleration change".
Imagine a "max speed test" where you fly almost horizontally, but with a strong pitch angle for 10+ seconds.
Releasing throttle at top speed reproducibly forced my quad into a tilted state that affected all axes.
This state was hardly controllable as the quad had a combined roll and pitch angle which i saw in the FPV. The internally leveled horizon seemed to be tilted. The stick commands worked "off-horizon"
The only solution was to gain hight and get home somehow. During half a minute of tilted cruising the horizon stabilized.

etracer65, 2018-05-01:
@stevo1957 It's likely that you're not keeping the stick in the transition zone that engages acro while in horizon mode.  In other words, you're moving the stick back towards center before the flip completes and then self-level is engaged and the quad tries to find the shortest route to level - which might be by rolling back to level.  I did a number of tests earlier today in LOS horizon mode and couldn't replicate any problems.

RipperDrone, 2018-05-01:
Maybe younwantomgive set iterm_rotation = On a try? There was a recent fix of I term being calculated in the wrong coordinate system (fixed vs onboard).

RipperDrone, 2018-05-01:
I tried Horizon lately, strange flip behavior confirmed. It's like doing a 180Yaw at the end of a fast flip. Might have to do with item calculation...? Or is it more likely about the transition @etracer65 has mentioned?

BTW, I have setpoint and transition enabled, too. Maybe they are causing time jitter of stick inputs, thereby confusing the transition from Horizon to Acro when doing flips?

stevo1957, 2018-05-01:
Thanks for the replies.

If it is an issue being experienced by others, then I can live with it. 

I was thinking it may be a fault with my controller but sounds like this is not the case.

AirBreak69, 2018-05-01:
Setting iterm_rotation = ON solved the "tilted horizon" problem - thanks! So ON should be the default for future releases.
Using this option the yaw shift after releasing sticks during incomplete flips seems to increase from 90 to 180 degrees.

AirBreak69, 2018-05-01:
No. Before using that option I only remember of 90 degrees yaw shifts (CCW direction) after incomplete flips.
With iterm_rotation = ON the yaw shifts today allways had 180 degrees (or 0 if the flips were complete enough).
But to be honest/human: Where is front, if you fall upside down?

stevo1957, 2018-05-05:
Thanks guys,

I am still to try the suggested changes but will do do next time out at my flying field.

closed 2018-06-11 by stale[bot]