betaflight/betaflight #5791: OSD Bug with OmnibusF4 and latest Beta

Vidalcris, 2018-05-01:
Hi,
With the latest beta #5790 i got weird bugs on my osd. (OMNIBUSF4)

"Disarmed" become "Dis rmed" and i got a "ESCCCCC" error message when armed.

I tried flashing this version again and the problem is still there.

No problems with 3.3.1

'''
# Betaflight / OMNIBUSF4 (OBF4) 3.4.0 May  1 2018 / 02:58:52 (8bb6d4292) MSP API: 1.39

# name
name BH130-Crisssss

# resources
resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 B14

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature LED_STRIP
feature AIRMODE
feature ESC_SENSOR
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -SYSTEM_INIT
beeper -ON_USB

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 16384 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# led
led 0 7,8::A:13
led 1 8,8::A:13

# color

# mode_color
mode_color 6 0 4
mode_color 6 1 13

# aux
aux 0 0 0 1700 2100 0
aux 1 13 2 1700 2100 0
aux 2 26 1 1700 2100 0
aux 3 35 2 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass2_type = BUTTERWORTH
set gyro_lowpass2_hz = 110
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_calibration = 59,145,-121
set mag_hardware = NONE
set baro_hardware = NONE
set rc_interp = MANUAL
set rc_interp_int = 12
set serialrx_provider = IBUS
set blackbox_p_ratio = 64
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT1200
set failsafe_throttle_low_delay = 10
set align_board_yaw = 90
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 33
set current_meter = NONE
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set tlm_inverted = ON
set osd_tim_2_pos = 2416
set osd_remaining_time_estimate_pos = 396
set osd_vtx_channel_pos = 2072
set osd_current_pos = 388
set osd_mah_drawn_pos = 402
set osd_craft_name_pos = 2048
set osd_avg_cell_voltage_pos = 2408
set osd_disarmed_pos = 2347
set osd_esc_tmp_pos = 2065
set osd_esc_rpm_pos = 2444
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set vtx_band = 5
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5658

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_notch_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set p_pitch = 22
set d_pitch = 14
set p_roll = 20
set i_roll = 45
set d_roll = 14
set p_yaw = 40

# rateprofile
rateprofile 0

set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50
set tpa_rate = 20

jflyper, 2018-05-01:
@Vidalcris 
#5790 is a PR for a new board support. Are you referring to something else?

Vidalcris, 2018-05-01:
Yeah sorry i mean the "beta" i'm using is just after this PR merged.
Nothing related to the PR itself :)
This is the version with the bug on my OSD : https://ci.betaflight.tech/job/Betaflight/797/

jflyper, 2018-05-01:
What if you do full-chip erase or do CLI `defaults`?

jflyper, 2018-05-01:
(Btw, use triple back quotes, not triple quotes to make a code section.)

Vidalcris, 2018-05-01:
Mhm i have not tried that, i will delete all the osd related settings in my diff and paste only that after flashing.
(Thanks for the tips.. sorry about that XD)

jflyper, 2018-05-01:
Ah, okay, the problem is reproducible. I think it's related to #5780, and thus not limited to Omnibus F4 (OMNIBUSF4 target).

etracer65, 2018-05-01:
Try #5792 to see if that fixes it for you.

etracer65, 2018-05-01:
I tested with an OmnibusF4 v3 and#5792 seems to fix it.

etracer65, 2018-05-01:
@Vidalcris Try this:

[betaflight_3.4.0_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/1964563/betaflight_3.4.0_OMNIBUSF4.hex.zip)

Vidalcris, 2018-05-01:
Working for me too, thanks a lot for the file ;)

Vidalcris, 2018-05-01:
Mhm i still got the weird ESCCCC warning :-/

jflyper, 2018-05-01:
I think that's over current warning from ESC telemetry...

etracer65, 2018-05-01:
Well you have ESC telemetry enabled and the ESC temp and RPM elements on the OSD screen.  So if you really don't have ESC telemetry connected or the ESC's aren't capable, turn it off.

However there does appear to be a small defaults bug with the `osd_esc_current_alarm` setting.  It defaults to 0 which means generate the alarm if the ESC current is >= 0 - that's why you're seeing the alarm.  You can fix this by `set osd_esc_current_alarm = -1`.  Or you can turn off ESC telemetry if you don't need it.

Vidalcris, 2018-05-01:
I never said i have no esc telemetry ;)
My sentence say what i mean, i use Temp and rpm. I dont receive any info about current.

I will try to set osd_esc_current_alarm to -1.
Thanks.

PR #5792 (2018-05-03): Avoid 0xFF character when writing to MAX7456 in auto-increment mode
Fixes #5780 
Fixes #5791 

The 0xFF character is used as an "escape" character by the MAX7456 when using auto-increment mode. The last byte of the characters used for the logo is the 0xFF character and this was causing the version and CMS instructions to not display.

Changed the logic to avoid the 0xFF character during the auto-increment phase and added a second pass if needed to update those characters in direct addressing mode.

Exposed (but not caused) by #5405.  Does not need to be applied to 3.3 as the problem began in 3 [... cut]
closed 2018-05-03 by mikeller