betaflight/betaflight #5794: Omnibus F7 BF3.4 #794 Nightly: Errational Yaw Twitching - was fine up to BF3.3.1

RipperDrone, 2018-05-01:
Tried so many settings to finally find out there is obviously a glitch in lates 3.4 Nightlies: Have flashed all my 10+ quads with BF3.4 #793/794 Nightly, and most of them run fine. However, on a SAM X220 with Omnibus F7 as FC, there are yaw twitches - not really following any pattern but several times a second when I do throttle or attitude changes. BF3.3.1 and 3.1.7 are running smoothly. Maybe related to timer glitches (?). Running DShot 600, but same thing happens with MultiShot. 

Here's the DIFF in case someone is interested... https://drive.google.com/open?id=1i71V6__FGlpk0UZlqe1JF1ktWMNMXZ2S

jflyper, 2018-05-03:
Can you check out pwm update freq on ESCs?

RipperDrone, 2018-05-05:
Tried tomswap btw BF 3.3.1 and 3.4 several times now. 3.4 = saw twitching. 3.3 = smooth

jflyper, 2018-05-06:
I have trouble flying a test Omnibus F7 (this one is V2) flashed with 794 with your config diff.

Can you start with the fresh config (`defaults`) and test if the symptom persist with minimalistic configuration; receiver, mode and motors. I did this and had no problem including yaw twitching on throttle you claim.

RipperDrone, 2018-05-06:
Have done some more testing.

ESCs are BLHeli_S, so no option for ESC frequency :-/:

[image]

RipperDrone, 2018-05-06:
Tried 3.4 Nightly #823 with defaults plus stepwise diff addition from 3.3. Guess what: All solid now. Maybe it had to do with the D term spikes which have been corrected in one recent commit? Closing this issue now, seems fixed.

closed 2018-05-06 by RipperDrone