betaflight/betaflight #5804: Spektrum reciever on Scisky F3 unusable

tenten8401, 2018-05-03:
I have one of these [F3 flight controllers](https://www.getfpv.com/oversky-scisky-micro-f3-brushed-flight-control-built-in-rx-option-dsmx-dsm2.html) and went through the [setup guide](http://www.overskyrc.com/index.php?main_page=fillib&file_path=F3FCfirmware%20FlashingGuide.pdf)

SBUS doesn't appear to work at all in that setup guide, and seeing how I'm using a DSMX transmitter on this, I am trying to use SPEKTRUM1024 & 2048 for serial providors, which are the only ones that move despite the guide telling me to use SBUS.

I have my ports setup as follows:
[image]

But whenever I move the sticks, this happens:
[image]

Here's a cli dump: https://hastebin.com/ugawejajuj.pas

I'm beginning to think it's a Betaflight bug, which is why I'm posting here. I'm using the SPRACINGF3EVO Betaflight preset, which the manufacturer's guide said does work, but they tested using an older version (3.1.6) so I'm thinking something may have changed somehow between there and 3.3.1

jflyper, 2018-05-03:
@tenten8401 
I have no trouble with RX with a generic F303CC test board flashed with v3.3.0 SPRACINGF3VO target firmware, loaded with your `dump` output. The only thing I had to change was the channel map, which I changed to "TAER1234".

[image]

I suspect you have a hardware problem, the board and/or the receiver.
Another theory is that your board is NOT compatible with SPRacingF3EVO.

closed 2018-05-03 by jflyper

tenten8401, 2018-05-03:
Yeah, I was afraid of that. I'll see if I can still return this board and pick up a different one.

AndersHoglund, 2018-05-03:
This is a typical miss-match of SPEKTRUM1024 and SPEKTRUM2048.  Rx in one of the modes and BFC in the other

jflyper, 2018-05-03:
@AndersHoglund Yes, that is another possibility. @tenten8401 You should check it out.

tenten8401, 2018-05-03:
I've tried both 1024 and 2048 and there doesn't seem to be a difference at
all, just more channels that glitch out when I move the sticks.

AndersHoglund, 2018-05-03:
Did you use a bind value of 9 (DSMX 2048/11mS) ? Then provider SPEKTRUM2048 should work.

tenten8401, 2018-05-03:
I've been using DSMX 22ms because that's what my transmitter seems to be
picking it up as, should I try switching the transmitter to 11ms?

tenten8401, 2018-05-03:
Also, can you elaborate on the "bind value" thing? Is it a setting in the
Betaflight configurator? Currently I've got it set to Spektrum2048 Serial,
I don't know if that means it's set to 9 or if I need to go change
something manually.

jflyper, 2018-05-03:
> Also, can you elaborate on the "bind value" thing?

It's a value (a number of pulses) you set to CLI variable `spektrum_sat_bind` to use with spektrum sat bind facility of BF.
https://github.com/betaflight/betaflight/wiki/Spektrum-Satellite-Bind-for-3.2

tenten8401, 2018-05-03:
I'm not sure if it uses an "official" Spektrum Satellite receiver, so I
don't know if that'll actually make a difference, however I'll definitely
give it a shot when I get home.

AndersHoglund, 2018-05-03:
Let me re-phrase the Q, How did you bind Rx to Tx ?
Via cli you can "set spektrum_sat_bind = 9" and save. There are other values too, as per the WiKi:
https://github.com/betaflight/betaflight/wiki/Spektrum-Satellite-Bind-for-3.2

AndersHoglund, 2018-05-03:
Now I find that it has Auto-bind. Wonder if it can also do forced binding from the FC? 
I also find this on the getfpv.com site:

> Every built-in receiver on the board outputs receiver signal in the same 8CH PPM format and order, the channel number depends on customer’s transmitter

A bit contradicting to what the setup guide says. Is this a Serail Rx or a PPM Rx ?

tenten8401, 2018-05-03:
I literally just turn it on, wait for it to start blinking after not
receiving a signal, and bind the controller to it.

tenten8401, 2018-05-03:
Seems to be serial as far as I can tell, PPM doesn't seem to do anything.

AndersHoglund, 2018-05-03:
You might get more help on RCG, There is one post suggesting that it IS a PPM Rx. 
https://www.rcgroups.com/forums/showpost.php?p=36719938&postcount=12
A lot of confusion. Get another FC

tenten8401, 2018-05-03:
Think I figured out why some people are saying it's PPM. The Hubsan and
FlySky controllers output PPM, while the DSMX/FrSky ones output through
Serial. I'll try that Spektrum bind trick out and get back to y'all, but if
it doesn't work I'm probably just going to return this board.

astragalizont, 2018-05-25:
i've got teh same problem on a Matek F405, binding with provider 9 and 2048-11ms results in no connection, with 2048-22 wierd glitching as in OP. It is an spm4649t

jflyper, 2018-05-25:
Cli diff please.

Rascalgp, 2018-08-01:
I have the exact same problem as @tenten8401 

Did you end up figuring it out?

I use a Spektrum DSM2 DX5e transmitter

tenten8401, 2018-08-01:
@Rascalgp Since posting this I decided to order another board and it works as expected. I have no clue what the issue was.
https://www.banggood.com/Eachine-32bits-F4-Brushed-Flight-Control-Board-Based-On-Openpilot-CC3D-Revolutio-For-Micro-FPV-Frame-p-1084264.html

Rascalgp, 2018-08-01:
Thanks for the reply, i guess i will try to return the product..