betaflight/betaflight #5814: Improve prop wash by dynamic throttle related increase in D

ctzsnooze, 2018-05-04:
Here's a challenge!  :-)

Most pilots can be very happy with filtering options coming out in 3.4, and the new throttle boost features.  One persistent 'flight' issue that is not entirely resolved is prop wash wobble.  

I do know from some careful testing that prop was is improved by more D and having the least possible delay in the input to response chain.  With the better D filtering in 3.4 I've had good results pushing D higher than I could before, even with quite hard D filtering, while filtering gyro a bit less.

The delay component primarily is mostly in the motor/prop combination because it just takes time for the motors to speed up.  There are relatively much smaller delays in the filtering, computing and other FC code elements.  Filter delay in 3.4 with dual PT1's and fixed notches removed isn't a big factor, and in any case it is unavoidable.

Most prop wash is around 30Hz.  That means the direction of the gyro signal reverses every 15ms, so our gyro to thrust response time has to be really fast; if delayed, the responses get out of phase and make things worse, not better.  

Looking closely at how the PID system responds, it's obvious that only D generates a response quick enough to do any good.  P is much less helpful, lagging behind quite a lot.  This explains why higher D is more effective on prop wash than higher P.

Prop wash typically happens when we throttle up while falling backwards, and hangs around for a short time until we get into clean air or start moving forwards.  

So I was thinking that a possible solution is to use methodology similar to anti-gravity, modified to detect only positive changes (differential) in throttle.  Put that through a smoothing low-pass filter, so that we get a smooth positive 'lump' of a signal every time we throttle up - nothing when we throttle down, nothing at constant throttle.  

Then use this to multiply D by some amount (maybe max of twice as much).  As we throttled up, D would help stabilise the quad against wobble, smoothly increasing around that time and then dropping back to normal.

I am fairly confident that this would assist significantly in controlling prop wash, and not otherwise do much in the way of bad things.

Anyway the coding side is a bit too complex for me, I'm putting the idea here in case someone can have a go at it.  Very happy to build and test.

This graphic shows why D, rather than P, is a better driver to oppose prop wash:

[image]

The horizontal zoom is 1000%, so we are looking at 100ms of signal. Note how once the gyro trace starts to dip down, P remains the wrong way until the gyro trace reaches zero, only D responds quickly, before the gyro trace reaches zero.  Because of P delay, it takes about 5ms before PID_Sum gets the correct 'sign' and start telling the motors the right thing to do.  If D was stronger, that delay would be less.  The implication is that increasing D, and possibly reducing P, are the main PID changes that would deal with prop wash quicker and more effectively.  Probably, doing so would reduce amplitude and increase frequency of prop wash, not eliminate it, but the overall amplitude reduction would probably be very useful

theSkonk, 2018-05-04:
Could you maybe use the accelerometer to detect the conditions that lead to prop wash ( moving backwards and falling at the same time ) to trigger the D multiplier?

flint723, 2018-05-04:
There was some discussion about accel on slack last night, I think the answer is accel isn't reliable enough to know what is "down" in acro flying.  I think Codecae was going to look into it a little bit.

In theory if you were falling backwards at an angle just right with props idling, vertical (in relation to quad frame) accel would be +1.xx (because of prop drag), with left/right and forward/back accel reading zero.  This is identical to a hover with very slight throttle increase.

I could be wrong about this, I don't claim to understand completely how the accel works.

codecae, 2018-05-05:
Accel will only really determine orientation, not direction of movement.  Since backward falling orientation can look the same as forward flight, there's not really much of a benefit in using it.

closed 2018-06-11 by stale[bot]