betaflight/betaflight #5826: Omnibus F4 emax bullet 30a dshot unresponsive

mmakam2, 2018-05-05:
DYS F4 AIO Omnibus F4 target on Betaflight 3.3.1. ESC's are Emax 30a Bullets DShot supported and on blheli. When arming the quad and giving a little throttle, the ESC's do start up beeps and then start erratically spinning in a pulsing fashion with no input changing it.

Attempted Fixes:

Disabled OSD
Dropped PID/Gyro to 2.7khz
Multishot works works perfectly 

# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack: https://slack.betaflight.tech or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

etracer65, 2018-05-05:
I have a couple of builds with these ESCs and have no problems.  Make sure you select DSHOT600 as 1200 is not supported.  Also make sure you've updated the ESCs to the latest BLHeli_S firmware.

mmakam2, 2018-05-05:
I have tried all Dshot versions, I have also updated the ESCs to the latest version.

What build Betaflight are you on?

mmakam2, 2018-05-05:
The ESCs are probably fine, I'm thinking its something to do with the the FC or betaflight build.

jflyper, 2018-05-05:
OMNIBUSF4 3.3.1 on Omnibus F4 AIO (gen1) runs Wraith32 x 4 without a problem, with all protocols: OS125, OS42, MS, DS150,300,600 and 1200.

closed 2018-05-05 by jflyper

mmakam2, 2018-05-05:
You know there are other people having this issue with simliar setups...

mmakam2, 2018-05-05:
The wiring is perfect, quad flies perfect, multishot works great. Why would Dshot not work if it was not the software?

jflyper, 2018-05-05:
Can we see CLI `diff` output from the configuration that doesn't give a good spin?

jflyper, 2018-05-05:
> The wiring is perfect, quad flies perfect, multishot works great. Why would Dshot not work if it was not the software?

We can not exclude that case that there is a problem with the ESCs. Have you successfully used those ESCs with DShot in any build/configuration?

mmakam2, 2018-05-05:
I have not used these specific ESCs, but ive used the same ESCs on different build with no issue.
CLI DUMP:

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.3.1 Apr  3 2018 / 09:54:44 (4b7eaf79d) MSP API: 1.37

# name
name 210R

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 A01
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 NONE
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
[... 616 lines cut ...]
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 105
set pitch_rc_rate = 105
set yaw_rc_rate = 105
set roll_expo = 5
set pitch_expo = 5
set yaw_expo = 0
set roll_srate = 71
set pitch_srate = 71
set yaw_srate = 70
set tpa_rate = 11
set tpa_breakpoint = 1620

etracer65, 2018-05-05:
A `diff` would have been more helpful. Have you remapped any of the motors (or anything else) using the `resource` command?  Can you also provide the output from the `dma` command (make sure DSHOT is selected first)?

mmakam2, 2018-05-05:
Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

jflyper, 2018-05-05:
The provided config dump works. Must be a issue with ESCs, FC and/or build itself.

mmakam2, 2018-05-05:
So i did remap one of the led resources for controlling a servo (camera angle), this issue existed before I did this setup though.

mmakam2, 2018-05-05:
If it was the ESCs, i doubt all 4 would be bad at once, so it must be the FC

mmakam2, 2018-05-05:
this is the exact FC i am running:

https://www.getfpv.com/dys-f4-flight-controller-poker-face.html

mmakam2, 2018-05-05:
Wow i see that there is a betaflight release for dysf4pro, did not realize this, it shipped with omnibus target. I will try this and report back.

etracer65, 2018-05-05:
Why aren't you running the DYSF4PRO target?

jflyper, 2018-05-05:
DYS F4 PRO does have it's own target, but not a big deal...

Too bad my test DYS F4 PRO has been broken without replacement for half a year now 😭

jflyper, 2018-05-05:
@etracer65 getfpv says it uses OMNIBUSF4...

etracer65, 2018-05-05:
Try using target DYSF4PRO.  There is a difference that may help as compared to OMNIBUSF4.

```
#if defined(OMNIBUSF4SD) || defined(DYSF4PRO)
#define ENABLE_DSHOT_DMAR       true
#endif
```

I doubt getfpv updated their product page after the original product was offered.

jflyper, 2018-05-05:
I would check Blheli versions and such.

mmakam2, 2018-05-05:
exact same issue with DYSF4PRO target, I am on BLHeli 16.7 for the ESCs

mmakam2, 2018-05-05:
Not sure if it matters but I have a 1000uf cap on the xt60 pads and a 220uf cap on the 5v circuit.

jflyper, 2018-05-05:
Do you have signal ground wired between FC and ESC?

mmakam2, 2018-05-05:
Yes I have signal+ground for all ESCs

jflyper, 2018-05-05:
I'm out of ideas (except for swapping with working dshot esc)... 😭

mmakam2, 2018-05-05:
I have some cheapo 32bit furibees in the mail coming soon, will try that and report back as well.

jflyper, 2018-05-05:
This one is even cheaper...
https://www.gearbest.com/esc/pp_719423.html

etracer65, 2018-05-05:
I think I see the problem in your configuration:

`set dshot_idle_value = 200`

This is way too low (corresponds to 2.0% in the configurator).  The default value is 450 (4.5%).  Most motors will not startup and spin reliably at 2% - even the default of 4.5% is sometimes too low.  Try resetting back to the default.

mmakam2, 2018-05-05:
tried it, not the problem, and ive had plenty of dshot builds at 1% with zero issues.

mmakam2, 2018-06-01:
FYI, figured out the issue. I had a custom motor mix due to using non standard motor pins from FC position. dshot couldn't handle the mmix, although multishot could. When i did motor mapping via resource instead of mmix, it resolved the issue.