betaflight/betaflight #5827: Connection Issue Betaflight 3.3.1/BLHeli Configurator Omnibus F3

regnets, 2018-05-05:
I just want to make you aware that there is an issue with Betaflight 3.3.1 when you use the BLHeli Standalone Configurator or the Chrome WebApp. 

Currently this issue is just occuring if you got an omnibus Board as far as i know. There are already three issues in the BL Heli Configurator Repo, however i think this is related to Betaflight, as i was able to connect to my Board using Betaflight 3.0.1.

Here are the issues from the BLHeli Configurator Repo:
https://github.com/blheli-configurator/blheli-configurator/issues/139
https://github.com/blheli-configurator/blheli-configurator/issues/131
https://github.com/blheli-configurator/blheli-configurator/issues/124

I also discovered that the connection between the Betaflight Configurator and the F3 Omnibus isn't reliable as it takes sometime more the 3-4 attempts before there is a successful connection to the Board.

Between these attempts i need to power loop the flight controller. The error i getting is that the serial port isn't available.

These is reproducible for all (about ten) Omnibus F3 Boards as well as for the F3 Omnibus Boards of my friends. This behaviour isn't connected to a single computer or operating system. I already tried different computer with different Windows Versions so far (Windows 10/8.1/7).

jflyper, 2018-05-06:
I have a counter example.

I have no trouble connecting to Typhoon S from Omnibus F3 AIO (V1).

From Chrome version (1.0.14)
```
2018-05-06 @ 11:05:20 -- Running - OS: MacOS, Chrome: 66.0.3359.139, Configurator: 1.0.14
2018-05-06 @ 11:05:24 -- Serial port successfully opened with ID: 3
2018-05-06 @ 11:05:24 -- MultiWii API version received - 1.37.0
2018-05-06 @ 11:05:24 -- Flight controller info, identifier: BTFL, version: 3.3.1
2018-05-06 @ 11:05:24 -- Running firmware released on: Apr 3 2018 09:54:28
2018-05-06 @ 11:05:24 -- Board: OMNI, version: 0
2018-05-06 @ 11:05:24 -- Unique device ID received - 0x2f003c5433570e20333536
2018-05-06 @ 11:05:26 -- Reading setup
2018-05-06 @ 11:05:28 -- Reading setup finished
2018-05-06 @ 11:06:40 -- Log file written
```

From standalone version (1.1.0)
```
2018-05-06 @ 11:24:46 -- Running - OS: MacOS, Chrome: 63.0.3239.108, Configurator: 1.1.0
2018-05-06 @ 11:24:53 -- Serial port successfully opened with ID: 1
2018-05-06 @ 11:24:53 -- MultiWii API version received - 1.37.0
2018-05-06 @ 11:24:53 -- Flight controller info, identifier: BTFL, version: 3.3.1
2018-05-06 @ 11:24:53 -- Running firmware released on: Apr 3 2018 09:54:28
2018-05-06 @ 11:24:53 -- Board: OMNI, version: 0
2018-05-06 @ 11:24:53 -- Unique device ID received - 0x2f003c5433570e20333536
2018-05-06 @ 11:24:55 -- Reading setup
2018-05-06 @ 11:24:57 -- Reading setup finished
```

[image]

martymclaughlin05, 2018-06-10:
Just wondering im having trouble with blheli  i connect to betaflight fine disconnect, i connect to blheli fine the problem starts when i read setup it finishes and then leaves me with a reading setuo finished i dont know a lot soo i just need some help thanks also it worked the other day also this is on two quads i have

closed 2018-07-17 by stale[bot]