betaflight/betaflight #5832: RunCam Split Mini controls only work when disarmed

andrejpodzimek, 2018-05-05:
Betaflight doesn't seem to send any camera commands while armed. My camera power button channel works fine while disarmed; it (indeed) starts and stops the recording. Once armed, the same channel input does nothing. Is there a reason why camera commands can't be issued (e.g.) in flight?

mikeller, 2018-05-06:
If I remember correctly this was done intentional, to prevent users from accidentally powering off the camera while flying FPV.

andrejpodzimek, 2018-05-06:
No matter what I do (while disarmed), I can never power off the camera using the serial input. The only way to power it off (with battery plugged in) is the physical power button on the PCB. So this protection may be doing more harm than good. I have 2 Splits v1 and one Split mini and all of them act this way. The “power button” channel seems to be a “power button without the long press” channel.

tenten8401, 2018-05-06:
Perhaps have a setting in the configurator to enable/disable accidential camera shutoff protection?

andrejpodzimek, 2018-05-06:
But can the camera be powered off using UART alone? I guess the answer is **no** — I’ve never seen that happen and I’ve had RunCam Splits since they first evolved back in Permian and Carboniferous.

andrejpodzimek, 2018-05-07:
In that case I believe this is a bug (or usability issue) in the driver that should be fixed. Such a limitation doesn’t make sense to me. There is nothing wrong about starting and stopping the recording in (FPV) flight — the analog video signal is always available, regardless recording status.

A typical use case: I’m in 100 meters of altitude and realize I didn’t start the recording. This should be easy to resolve with just a click of a button. Yet with these new limitations (new indeed; camera controls worked while armed ~a month ago) I’d have to land, disarm, switch on the camera… I don’t understand **why** this limitation applies.

It would make sense to block (only) the OSD menu while armed — you don’t want to get into the menu by mistake while flying. But starting/stopping the recording in flight should, IMHO, just work.

jflyper, 2018-05-07:
From the `rcdevice_cam.c` source file, `CAMERA2` box looks like a power button. If it is not working, then THAT is a bug.

andrejpodzimek, 2018-05-07:
It works like a power button in terms of starting / stopping the recording. It doesn't work like a power button in terms of switching the camera on / off (i.e., it doesn't respond to a long "press").

This^^^ has always been the case, since the first Split, and it's perfectly fine; nobody wants to power the camera off in flight. Which gets me back to the original point: It doesn't make sense to ban the use of the virtual power button when armed. It won't switch the camera off and therefore it's safe to use in flight.

mizuta1951, 2018-05-14:
In Runcam Split-2 you can accidently power off, if you turn into Wi-Fi mode with no Wi-Fi module. And if it happens you will lose your FPV.

andrejpodzimek, 2018-05-14:
The WiFi command / button should be definitely blocked in flight, because you *always* lose analog video in WiFi mode, no matter if you have a WiFi module or not. Therefore it’s not safe for flight.

This feature request is solely about the power command — that doesn’t power off the camera and can be used in flight (to stop/start video recording).

mikeller, 2018-05-25:
@azolyoung: Can you make a statement on what the current state of this is, and if there is something here that needs to be changed / fixed in Betaflight before the 3.4 release?

dennygame, 2018-06-04:
@mikeller Hi Michael, my name is Denny, I work in a camera solution company. 

We would like to make our new generation FPV camera to be compatible with Betaflight, for example, to make it can be controlled by Betaflight via UART(in the past we have products can be controlled by AT commands but we don't know about how Betaflight do it).

It will be great if we can discuss, my email address is denny@biteye.cn, I'm looking forward to hear from you!

Regards

azolyoung, 2018-06-23:
@andrejpodzimek @mikeller

Hi,

Thanks for the suggestion, and we know the turn on wifi while in flight is danger, the next firmware(>2.0.0) of rcsplit has change the wifi page logic, when the wifi on, rcsplit will keep the tvout, and just put a wifi indicator  to the right top of the tvout. 

I'm fixing a bug of rcsplit in bf3.3(#6196), and also will put this condition(avoid turn on wifi when arming) to the wifi button handler.

jinteckhan, 2018-08-02:
What about betaflight 3.4? I used camera power / mode button control via UART. It surely works on 3.3.x but in the same wiring, It does not work on betaflight 3.4.
I can set modes in the mode tab. It shows the change according to my transmitter switch.
But in real flight, even in disarmed, It does not work.
Runcam split 2 and split mini have the same problem.
Is there anyone who has the solution about this problem?

Xilllius, 2018-08-02:
Same here, tried 3.4 up tp rc5 and the controll doesnt work

joeostrander, 2018-08-07:
I'm not sure if this will help, but I had 2 RC Splits, a V1 and a V2.  Both worked fine, but when I upgraded BF on both quads, the V1 quit working (couldn't start/stop recording via switch).  I reverted BF and V1 started working again.  I upgraded the firmware on my V1 and it started working just fine with latest BF.

Pablofpv, 2018-08-19:
hi all, i have two runcams split 2 and the camera control trougth the uart with betaflight 3.4 and 3.5
with 3.3 all was fine before

Pablofpv, 2018-08-27:
It's there any solution for this issue?

mheberlein, 2018-09-16:
Looks like camera control isn't working at all with my setup, no matter whether it's armed or disarmed:
- RunCam Split v1, updated firmware to V2.0.0
- connected via SoftSerial TX1/RX1
- to BFF3 running Betaflight 3.5.1

I tried compiling the latest Betaflight with `#define USE_CAMERA_CONTROL` explicitly set but that didn't change anything.

closed 2026-09-26 by blckmn