betaflight/betaflight #5835: PIKOBLX - very stubborn to reset FC

RipperDrone, 2018-05-06:
HAve 2 PIKOBLX builds, and they both seem to be very resistant against reboots :-)

Whenever I try to reboot the FC after flashing or even manytimes during configuration, I have to unplug USB and plug it back in to make the FC reboot reliably. Seems it gets stuck in some mis-timing on the USB serial protocol very often...? :-/

mikeller, 2018-05-06:
@MJ666 would be the expert to ask about this.

MJ666, 2018-05-06:
@RipperDrone I can not confirm your problem on the PIKOBLX and some clones i have. What version of the firmware you are using? Are you using original PIKOBLX boards for FuriousFPV? What happens after reboot of the FC most of the time you need to reconnect with the BF configurator again manually after a moment. This is not very specific to the PIKOBLX and happens for many of the VCP targets but they always reboot properly.

RipperDrone, 2018-05-13:
@MJ666 Boards are both genuine FuriousFPV. Tried both official 3.3.1 and Nightly Build from beg of May.

RipperDrone, 2018-05-17:
@MJ666 Have a fast desktop computer, tried different USB ports + cables. Anyway, workaround is ok, it's just that all other FCs (I have >30 builds now) connect very reliably w/o any need to unplug USB/retry. Yes, in some cases restarting BFC helps, too...

closed 2018-06-23 by stale[bot]