betaflight/betaflight #5838: Mass Storage feature reads corrupted data

nullcode, 2018-05-06:
Using msc, blackbox logs are corrupted and cannot be read by the blackbox log viewer. After transferring the files with an SD card reader the logs work flawlessly.

firmware v3.4-823
using macOS

mikeller, 2018-05-25:
I think there still is a problem with MSC support under MacOS, @jflyper?

Also, @jflyper, did you want to have another look into cleaning up the MSC for onboard flash implementation, including deciding on keeping / ditching the colour icon, before we release 3.4?

eniefaC, 2018-09-23:
Spent so much time on this today before finding this open issue... I can reproduce on my Matek F722-STD with Betaflight 3.5.0 on Windows 10.

I've attached two files here ([LOG00001.zip](https://github.com/betaflight/betaflight/files/2409150/LOG00001.zip)):
- The first one is downloaded through the USB mass storage option in Betaflight. 
- The second one is downloaded through an SD card reader.

Interestingly the header for the mass storage file wrongly report that I'm on BF 3.4.0.

Note  : to any dev reading this: thanks for this wonderful piece of software 👍 !

mikeller, 2018-09-24:
@eniefaC: Is this the same log (on the board), downloaded twice in different ways? If so, it's probably caused by the log itself being corrupted when it was written.

eniefaC, 2018-09-24:
@mikeller Yes absolutely. It's the same file, just downloaded differently.

mikeller, 2018-09-24:
@eniefaC: Hmm, the log is working fine for me when loaded into blackbox-log-viewer 3.2.0.

eniefaC, 2018-09-24:
@mikeller Here's what I see : 
[image]

Left side is via mass storage, *wrongly* reporting BF 3.4.0, 3S lipo... Length is 00:01.302.
Right side is via card reader, *correctly* reporting BF 3.5.0, 4S lipo... Length is 00:17:810.

You can tell by the graph on the bottom that there is data for the card reader file, but none for the mass storage file. Both files are the exact same size.

Please tell me if there's anything else I can help with 😄 !

mikeller, 2018-09-24:
@eniefaC: Ok, I misread your initial report, and (by coincidence) I checked only the SD card reader version. I have downloaded logs from SD card to linux over the Betaflight MSC driver in the past, and it's always worked fine, so I am wondering what is going on here.

dabit20, 2018-12-29:
Today I ran into the same issue using Betaflight 4.0 build 1315 on a Matek F405STD which uses an SD card for logging.

Log files are correctly read when ejecting the SD card and putting it into an external reader. When putting the FC in MSC mode and extracting the files that way yields corrupted files. 
I tried to read the files with a fresh git pull of the Blackbox Explorer, maybe the file format has changed. But no, files are still corrupt.

I used MSC mode before with BF3.5.2 and as far as I can remember there were no issues.

PC is running Windows 10.

Attached are two versions of the same log, one read through MSC, one through the card reader. 

[LOG00145.zip](https://github.com/betaflight/betaflight/files/2716779/LOG00145.zip)

JaceRyczak, 2019-07-04:
I have also just encountered this issue on my MATEK F722-SE running BF 4.03.

Files transfer fine off the SD card using Mass Storage mode, however the BBE software fails to read the logs as they cannot be "parsed" correctly.

Attached are two files, they are the same flight log, except one was downloaded using Mass Storage Mode and the other by removing the SD card and using a reader. the failed log reports an incorrect version of BF as mentioned above also.

Is this problem still being worked on?
[Desktop.zip](https://github.com/betaflight/betaflight/files/3358375/Desktop.zip)

TimCiarlette, 2019-07-23:
I am having issues with Mass Storage mode on Matek F722-SE also, currently on BF 4.04, some times it works and other times it doesn't. 

Files transfer fine off the SD card using Mass Storage mode, however the BBE software fails to read the logs as they cannot be "parsed" correctly.

I will be removing the SD card for reading until issue fixed.

Also: Thank You to all the Betaflight Team Members.

mikeller, 2019-07-25:
@TimCiarlette: A lot of micro SD cards have problems with the way most boards access them. Try different cards, and different brands, to work out what is working reliably.

azixaka, 2019-08-12:
I am running BF 4.05 on Matek F722-SE and getting exactly the same issue as described by eniefaC. MSC is very flaky and only 3 out of 10 files are transferred correctly the rest are simply corrupt and can't be parsed by BBE or have no throttle chart at the bottom. If I take out the SD card and copy these files using SD card reader they are fine. Tried Sandisk Extreme Plus 32GB, Samsung Evo Select 32GB, Lexar 633x 16GB and Sandisk Ultra 16GB. Tried formatting using SD Card Formatter as well. All on Windows 10 1903 and all SD cards are FAT32.

Could it be something to do with SD card sharing resources/DMA with some of the UARTs as I use them all.

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis