betaflight/betaflight #5839: BF3.4-823 1 motor spins to 100% after flip or roll

nullcode, 2018-05-06:
Tried to filter tune, but after a roll or flip one of the motors (not the same) consistently spun up to 100% resulting in a crash. Switching back to BF3.3.2 the problem did not occur anymore.
I have blackbox logs to prove this.

BF3.4-823, 32/16/16, pid and filters stock (only notches disabled).

etracer65, 2018-05-06:
Please provide a `diff` as requested in the instructions.

etracer65, 2018-05-06:
Still need to see a `diff` of your settings, but a couple of observations.

Running 32/16, DSHOT1200, and with the accelerometer enabled your CPU load is likely very high - possibly too high.

In the first 2 logs motor 3 appears to have stopped producing thrust.  Could be a motor or ESC problem.

The third log is less definitive and there are a couple of unexplained apparent loss of thrust events for motor 3 again.  But at the end of the log the craft appears to be under control and the gyros are near zero so I'm not sure where the problem is.

etracer65, 2018-05-08:
I would try lowering your CPU usage to rule that out.  Maybe go with 16/16 and disable the accelerometer.  You could also drop back to DSHOT600 to rule out problems with DSHOT1200.

nullcode, 2018-05-22:
@etracer65 thanks for the tips, at the end it was the ESC, but unfortunately I found it out in a bad way, went out to fly and after 5 seconds in the flight i did a roll and the quad dropped out of the air and the ESC started burning. It's very weird that a brand new DYS Aria 32 ESC did this.
I will close this since it's definitely not a BF 3.4 issue.
Thanks again.

closed 2018-05-22 by nullcode