betaflight/betaflight #5843: Universal target todos

jflyper, 2018-05-07:
- [ ] Timer selection index is ... zero origin?
- [ ] Printed timer channel number is zero origin.
- [ ] Acc/gyro should be configurable.
- [ ] Serial reports non-configured (pin-less) UARTs.

- [ ] Config dependent code embedded in `fc_init.c`.
https://github.com/betaflight/betaflight/blob/851ed5f5977cd3f928f5793f90e2e7e90e40e9ca/src/main/fc/fc_init.c#L516-L528

mikeller, 2018-05-07:
I think the EXTI pins need to become configurable as well for a number of devices.

And then there is 'exotic' hardware like SPI RX that need more extra pins...

jflyper, 2018-05-14:
- [ ] Decide which CSN pin to initialize at runtime (`spiPreInit`).

mikeller, 2018-06-20:
Pinned it.

closed 2026-09-26 by blckmn